Woutrrr/Omnik-Data-Logger #17: Jip en Janneke installatie

Ard1976, 2015-06-12:
Beste Woutrrr,

Ik ben bezig om jouw programma werkend te krijgen. Ben geen programmeur en ben nu aan het stoeien met Putty. Is er ook een Jip en Janneke versie van jouw stappenplan?

Thanks.

Ard

Woutrrr, 2015-06-16:
Helaas, er is geen Jip en Janneke versie beschikbaar. Maar wat heb je tot nu toe gedaan en waar loop je precies op vast?

Ard1976, 2015-06-19:
Beste Wouter,

Ik heb helemaal geen git ervaring. Met google toch een stuk verder gekomen.
Ik heb Mysql en PHPmyadmin geinstalleerd op mijn Qnap. Ik wil de Qnap gebruiken om daar het script op te draaien.
Ik heb de gitclone uitgevoerd naar mijn laptop en daar de config.cfg file aangepast. Ik neem aan dat deze config.cfg file op de Qnap geplaatst moet worden. In welke directory op de Qnap moet deze file geplaatst worden? Met welk commando via putty kan ik dit doen?

Met vriendelijke groet,

Ard Hermsen

From: Wouter van der Zwan 
Sent: Tuesday, June 16, 2015 11:01 PM
To: Woutrrr/Omnik-Data-Logger 
Cc: Ard1976 
Subject: Re: [Omnik-Data-Logger] Jip en Janneke installatie (#17)

Helaas, er is geen Jip en Janneke versie beschikbaar. Maar wat heb je tot nu toe gedaan en waar loop je precies op vast?

DiedB, 2015-06-19:
Download de hele map van dit project als een zip. https://github.com/Woutrrr/Omnik-Data-Logger/archive/master.zip

Download WinSCP, dat is voor een leek wat makkelijker dan de commandline wat betreft file management. Je kunt dan gewoon inloggen met de gebruikersnaam en het wachtwoord dat je bij Putty (SSH) gebruikt.

Kopieer de bestanden naar een map op je QNap (in /opt/bin/omnik bijvoorbeeld) en pas nu met WinSCP de configuratie aan.

Je kunt, als alles goed is gegaan, met het commando `python OmnikExport.py` het script draaien. De output komt dan op PVOutput (en in je MySQL database). Het gaat waarschijnlijk de eerste keer niet goed, zorg dat logging aan staat in de configuratie.

De volgende stap is zorgen dat het script elke minuut draait met cron, maar probeer eerst dit maar eens werkend te krijgen.

Ard1976, 2015-06-19:
Beste Diederik,

dank voor je input, WinSCP werkt inderdaad veel handiger voor een leek.
Bestanden gekopieerd naar de Qnap. Instelling op pv output verzorgd en config file aangepast.

Vervolgens commando uitgevoerd.
Ik krijg nu de volgende foutmelding:

Heb je een idee?

Met vriendelijke groet,

Ard Hermsen

From: Diederik Bakker 
Sent: Friday, June 19, 2015 8:08 PM
To: Woutrrr/Omnik-Data-Logger 
Cc: Ard1976 
Subject: Re: [Omnik-Data-Logger] Jip en Janneke installatie (#17)

Download de hele map van dit project als een zip. https://github.com/Woutrrr/Omnik-Data-Logger/archive/master.zip

Download WinSCP, dat is voor een leek wat makkelijker dan de commandline wat betreft file management. Je kunt dan gewoon inloggen met de gebruikersnaam en het wachtwoord dat je bij Putty (SSH) gebruikt.

Kopieer de bestanden naar een map op je QNap (in /opt/bin/omnik bijvoorbeeld) en pas nu met WinSCP de configuratie aan.

Je kunt, als alles goed is gegaan, met het commando python OmnikExport.py het script draaien. De output komt dan op PVOutput (en in je MySQL database). Het gaat waarschijnlijk de eerste keer niet goed, zorg dat logging aan staat in de configuratie.

De volgende stap is zorgen dat het script elke minuut draait met cron, maar probeer eerst dit maar eens werkend te krijgen.

DiedB, 2015-06-19:
Ik kan de foutmelding niet zien, is er iets fout gegaan met kopiëren?

Ard1976, 2015-06-19:
in de herkansing

/share/MD0_DATA/.qpkg/Optware/bin/Omnik/Omnik-Data-Logger$ python OmnikExport.py 
Traceback (most recent call last):
  File "OmnikExport.py", line 21, in <module>
    ip              = config.get('inverter','ip')
  File "/share/MD0_DATA/.qpkg/Python/lib/python2.7/ConfigParser.py", line 322, in get
    raise NoSectionError(section)
ConfigParser.NoSectionError: No section: 'inverter'

From: Diederik Bakker 
Sent: Friday, June 19, 2015 8:50 PM
To: Woutrrr/Omnik-Data-Logger 
Cc: Ard1976 
Subject: Re: [Omnik-Data-Logger] Jip en Janneke installatie (#17)

Ik kan de foutmelding niet zien, is er iets fout gegaan met kopiëren?

DiedB, 2015-06-19:
Lijkt erop dat je config-bestand niet kan worden ingelezen, want het loopt stuk bij het uitlezen van de eerste regel. Je config-bestand hoort config.cfg te heten en in dezelfde map te staan als OmnikExport.py.

Ard1976, 2015-06-19:
Ik heb bestandsnaam aangepast van config-org.cfg naar config.cfg. Vervolgens commando uitgevoerd en ik krijg geen foutmelding meer.
Ik zie nog geen data binnen komen bij pv output. Waar zou dit aan kunnen liggen?
Hoe kan ik kijken in de SQL of daar wel data in wordt geplaatst?

From: Diederik Bakker 
Sent: Friday, June 19, 2015 8:59 PM
To: Woutrrr/Omnik-Data-Logger 
Cc: Ard1976 
Subject: Re: [Omnik-Data-Logger] Jip en Janneke installatie (#17)

Lijkt erop dat je config-bestand niet kan worden ingelezen, want het loopt stuk bij het uitlezen van de eerste regel. Je config-bestand hoort config.cfg te heten en in dezelfde map te staan als OmnikExport.py.

DiedB, 2015-06-19:
Heb je in de config `log_enabled = true` gezet? Je kunt dan in het bestand logfile.log (in dezelfde map) zien wat er precies mis gaat. Paste het hier maar even.

Ard1976, 2015-06-19:
config staat inderdaad op true
log laat volgende zien

2015-06-19 21:02:12,998 INFO connecting to xxx.xxx.xxx.xxx port 8899
2015-06-19 21:02:23,024 ERROR could not open socket

ik heb ip adres van de omvormer even met xxx weergegeven. ip adres heb ik van het scherm van de omnik gehaald. Zou dus goed moeten zijn

From: Diederik Bakker 
Sent: Friday, June 19, 2015 9:18 PM
To: Woutrrr/Omnik-Data-Logger 
Cc: Ard1976 
Subject: Re: [Omnik-Data-Logger] Jip en Janneke installatie (#17)

Heb je in de config log_enabled = true gezet? Je kunt dan in het bestand logfile.log (in dezelfde map) zien wat er precies mis gaat. Paste het hier maar even.

DiedB, 2015-06-19:
Ik vrees dat ik toch het IP-adres even moet zien. Is het een adres binnen je interne netwerk (begint waarschijnlijk met 192.168) of staat er 10.10.100.254? In het laatste geval moet de WiFi-module eerst worden ingesteld zodat er verbinding is met jouw netwerk.

Ard1976, 2015-06-19:
is inderdaad een 192.168 adres

From: Diederik Bakker 
Sent: Friday, June 19, 2015 9:28 PM
To: Woutrrr/Omnik-Data-Logger 
Cc: Ard1976 
Subject: Re: [Omnik-Data-Logger] Jip en Janneke installatie (#17)

Ik vrees dat ik toch het IP-adres even moet zien. Is het een adres binnen je interne netwerk (begint waarschijnlijk met 192.168) of staat er 10.10.100.254? In het laatste geval moet de WiFi-module eerst worden ingesteld zodat er verbinding is met jouw netwerk.

DiedB, 2015-06-19:
Kun je vanaf de Qnap je omvormer pingen? Op de commandline van de Qnap het volgende commando:

`ping 192.168.*.*` waarbij de sterretjes de rest van het adres voorstellen.

Wat is de output?
`** bytes from 192.168.*.*` of iets anders?

Ard1976, 2015-06-19:
Ik krijg als terugmelding dat de host niet meer communiceert sinds 15 seconden

From: Diederik Bakker 
Sent: Friday, June 19, 2015 9:33 PM
To: Woutrrr/Omnik-Data-Logger 
Cc: Ard1976 
Subject: Re: [Omnik-Data-Logger] Jip en Janneke installatie (#17)

Kun je vanaf de Qnap je omvormer pingen? Op de commandline van de Qnap het volgende commando:

ping 192.168.*.* waarbij de sterretjes de rest van het adres voorstellen.

Wat is de output?
** bytes from 192.168.*.* of iets anders?

DiedB, 2015-06-19:
Heb je de module verbonden met het Omnik-portal, en werkt dat wel?

Wat is het resultaat als je `ping google.nl` probeert?

Ard1976, 2015-06-19:
De omvormer is verbonden met de Omnik-portal en daar komt info binnen.
ping google.nl geeft inderdaad wel reactie.
De poort instelling die in de config staat is een standaard poort?

From: Diederik Bakker 
Sent: Friday, June 19, 2015 9:44 PM
To: Woutrrr/Omnik-Data-Logger 
Cc: Ard1976 
Subject: Re: [Omnik-Data-Logger] Jip en Janneke installatie (#17)

Heb je de module verbonden met het Omnik-portal, en werkt dat wel?

Wat is het resultaat als je ping google.nl probeert?

Woutrrr, 2015-06-19:
Dan lijkt het er toch op dat het ip niet helemaal goed is. Hoe heb je dat ip gevonden? Via je router zou je het moeten kunnen vinden, of via de omnik smartphone app kan het ook.

 In de app voordat je inlogt in de app, kan je op manual mode klikken. Vervolgens krijg je een lijstje met omvormers te zien met de ip adressen. 

Je kan je instellingen testen via het LiveStats.py script. Als je alle settings in je config.cfg file hebt gezet, kan je LiveStat.py op je Qnap starten via de commandline. Ga daarvoor naar de map met de Omnik-Data-Logger bestanden en voer `python LiveStats.py` uit.

Woutrrr, 2015-06-19:
Port 8899 is inderdaad de standaard port om data bij de Omnik omvormer op te vragen.

Ard1976, 2015-06-20:
Beste Wouter en Diederik,

Ik ben weer een aantal stappen verder.
Het bleek dat het ip adres van de Omnik omvormer wel juist was. Ik had de Omnik echter in een gasten netwerk hangen. Hierdoor was hij niet zichtbaar voor de Qnap in het reguliere netwerk waar ook de Qnap in hangt. Ik heb de Omnik nu ook in het reguliere netwerk gehangen en kan hem pingen. Ook LiveStats.py werkt nu.
Volgende uitdating is de OmnikExport.py. Als ik deze uitvoer, dan krijg ik de volgende foutmelding:

/share/MD0_DATA/.qpkg/Optware/bin/Omnik/Omnik-Data-Logger$ python OmnikExport.py
Traceback (most recent call last):
  File "OmnikExport.py", line 85, in <module>
    import MySQLdb as mdb   
ImportError: No module named MySQLdb

Ik heb het volgende gedaan:

1. Ik heb in de config.cfg gewijzigd van 127.0.0.1 naar mysql_host = xxx.xxx.x.x (IP adres Qnap)
2. Ik heb via PhpMyadmin een database aangemaakt in MYSQL met de naam Omnik
3. Ik heb de toegang voor de MYSQL database (nog) niet gewijzigd, dus heb in config.cfg de volgende waarden ingevuld 
mysql_user = root  
mysql_pass = admin 

Hebben jullie een idee met betrekking tot de foutmelding?

Met vriendelijke groet,

Ard Hermsen
From: Wouter van der Zwan 
Sent: Friday, June 19, 2015 9:48 PM
To: Woutrrr/Omnik-Data-Logger 
Cc: Ard1976 
Subject: Re: [Omnik-Data-Logger] Jip en Janneke installatie (#17)

Dan lijkt het er toch op dat het ip niet helemaal goed is. Hoe heb je dat ip gevonden? Via je router zou je het moeten kunnen vinden, of via de omnik smartphone app kan het ook.

In de app voordat je inlogt in de app, kan je op manual mode klikken. Vervolgens krijg je een lijstje met omvormers te zien met de ip adressen. 

Je kan je instellingen testen via het LiveStats.py script. Als je alle settings in je config.cfg file hebt gezet, kan je LiveStat.py op je Qnap starten via de commandline. Ga daarvoor naar de map met de Omnik-Data-Logger bestanden en voer python LiveStats.py uit.

DiedB, 2015-06-20:
Python heeft (standaard) nog geen MySQL-support. Je hebt hiervoor de package python-mysqldb nodig. Ik weet zo snel niet hoe je dat op een Qnap kunt installeren.

Kun je eerst eens testen door MySQL uit te zetten in de config-file (false in plaats van true)? Komt er data op PVOutput?

Ard1976, 2015-06-20:
Ik heb MySQL op false gezet. Vervolges python OmnikExport.py uitgevoerd (ik krijg dit scherm /share/MD0_DATA/.qpkg/Optware/bin/Omnik/Omnik-Data-Logger$ python OmnikExport.py).
Geen vervolg en er komt geen data binnen op pvoutput.org.

From: Diederik Bakker 
Sent: Saturday, June 20, 2015 3:37 PM
To: Woutrrr/Omnik-Data-Logger 
Cc: Ard1976 
Subject: Re: [Omnik-Data-Logger] Jip en Janneke installatie (#17)

Python heeft (standaard) nog geen MySQL-support. Je hebt hiervoor de package python-mysqldb nodig. Ik weet zo snel niet hoe je dat op een Qnap kunt installeren.

Kun je eerst eens testen door MySQL uit te zetten in de config-file (false in plaats van true)? Komt er data op PVOutput?

DiedB, 2015-06-20:
Geen errors in de logfile?

Ard1976, 2015-06-20:
nee geen error.
2015-06-20 16:28:53,099 INFO connecting to 192.168.X.XXX port 8899
2015-06-20 16:28:53,755 INFO ID: NLDNxxxxxxx

Klopt het dat er geconnect wordt met de omvormer?

From: Diederik Bakker 
Sent: Saturday, June 20, 2015 4:26 PM
To: Woutrrr/Omnik-Data-Logger 
Cc: Ard1976 
Subject: Re: [Omnik-Data-Logger] Jip en Janneke installatie (#17)

Geen errors in de logfile?

DiedB, 2015-06-21:
Heb je PVOutput wel enabled? Je zou namelijk direct een message in de log moeten krijgen zodra het uploadproces start.

Ard1976, 2015-06-21:
PVOutput was inderdaad enabled. Ik heb alles nog eens nagelopen en alles stond goed. Vervolgens Qnap maar eens herstart. Nu loopt alles wel 
Ik heb crontab aangemaakt en nu wordt om de minuut data verzameld en 1x per 5 minuten naar PVOutput verzonden. 
Vervolg uitdaging wordt om de MySQL functie toch draaiende te krijgen op de Qnap en om data vanuit de Slimme Meter te uploaden naar PVOutput. We hebben cirkel dan rond volgens mij.
Diederik en Wouter hartelijk dank voor jullie support.

From: Diederik Bakker 
Sent: Sunday, June 21, 2015 9:38 AM
To: Woutrrr/Omnik-Data-Logger 
Cc: Ard1976 
Subject: Re: [Omnik-Data-Logger] Jip en Janneke installatie (#17)

Heb je PVOutput wel enabled? Je zou namelijk direct een message in de log moeten krijgen zodra het uploadproces start.

closed 2015-12-08 by Woutrrr

hlubach, 2020-05-23:
Traceback (most recent call last):
  File "LiveStats.py", line 7, in <module>
    import OmnikExport
  File "C:\Users\hluba\Downloads\Omnik\OmnikExport.py", line 11, in <module>
    import ConfigParser
ModuleNotFoundError: No module named 'ConfigParser'

probeer het v ia Windows 10 werkend te krijgen , maar krijg deze melding
Python 3,8,3

drksavant, 2020-07-28:
Leuk Verhaal , ik dacht ik probeer dit ook eens , alles gaat goed tot de remote mysql database

de connectie werkt want openhaal zit er ook in

020-07-28 14:21:26,392 DEBUG Connect to database
Traceback (most recent call last):
  File "OmnikExport.py", line 170, in <module>
    omnik_exporter.run()
  File "OmnikExport.py", line 82, in run
    plugin.process_message(msg)
  File "/home/pi/OmnikToo/outputs/MysqlOutput.py", line 22, in process_message
    with con:
AttributeError: __exit__