Woutrrr/Omnik-Data-Logger #21: No output to PVout.org

Bswanny, 2015-12-23:
Started today with a fresh install. Got LiveStats working with the comments of Molekuul. 
When running OmniExport the following error occured:

Traceback (most recent call last):
  File "OmnikExport.py", line 132, in <module>
    OmnikExport.run()
  File "OmnikExport.py", line 49, in run
    'filename': mydir + '/' + config.get('log', 'filename'),
  File "/usr/lib/python2.7/ConfigParser.py", line 340, in get
    raise NoOptionError(option, section)
ConfigParser.NoOptionError: No option 'filename' in section: 'log'

I changed filename also to log_filename like in LiveStats and the error changed:

2015-12-23 14:25:20,976 DEBUG Importing output plugin 
Traceback (most recent call last):
  File "OmnikExport.py", line 132, in <module>
    OmnikExport.run()
  File "OmnikExport.py", line 72, in run
    __import__(plugin_name)
ValueError: Empty module name

Any help in solving this is appreciated.

Woutrrr, 2015-12-23:
When I was updating the script, I forgot to change the LiveStats.py. I'm
working on update which fixes the LiveStats.py.

The changes you made to get the LiveStats.py working,  reversed the changes
I made for the new version of omnik export.

Tomorrow or even this evening I will upload the new version.

Bswanny, 2015-12-23:
Dank voor het antwoord,

Was even vergeten dat ik hier gewoon nederlands  mag gebruiken.

Met vriendelijke groet,

Bart

Woutrrr, 2015-12-23:
The new version is online, which should fix the LiveStats script.
Can you try the latest version and see if it works for you?

Bswanny, 2015-12-24:
Hi Wouter,
Just made a clean install. After changing the settings in Config.cfg, LiveStats works fine. OmnikExport returns the following error:
2015-12-24 10:21:32,271 DEBUG Importing output plugin 
Traceback (most recent call last):
  File "OmnikExport.py", line 170, in <module>
    omnik_exporter.run()
  File "OmnikExport.py", line 52, in run
    __import__(plugin_name)
ValueError: Empty module name

I noticed that in config.cfg the plugin PVoutputOutput was not enabled and added this.
Now i don't have any errors, but also nothing at pvout.org
After trying several times (5 minute interval at PVout.org) i got the error:

" ConfigParser.NoOptionError: No option 'apikey' in section: 'pvout' "

Looking in Config.cfg i noticed that the options are: pvout_apikey and pvout_sysid. 
Removing the pvout_ gets it working.

Hope this is helpful for others.

Have a nice Christmas

Bswanny, 2015-12-24:
Hi Wouter,

When running OmnikEport manualy everthing works fine. With the cronjob however i get the following error:
2015-12-24 15:05:01,806 DEBUG Importing output plugin PVoutputOutput
Traceback (most recent call last):
  File "/home/pi/Omnik-Data-Logger/OmnikExport.py", line 170, in <module>
    omnik_exporter.run()
  File "/home/pi/Omnik-Data-Logger/OmnikExport.py", line 52, in run
    __import__(plugin_name)
ImportError: No module named PVoutputOutput

I know it has something to do with the location where the module is stored, but i don't know how to get this working.

For now i got it working by copying the PVoutputOutput.py into the Omnik-Data-Logger directory.

ZonPV, 2016-01-01:
Ook mijn clean install bleek geen foutmeldingen te geneneren in het bestand omnik-export.log. Maar data kwam niet binnen bij PVoutput.org.
De voorgestelde aanpassingen doorgevoerd aangaande in
config.cfg weghalen van de phrase: pvout_ 
Hopelijk is er morgen genoeg zon om omvormer te laten starten en te testen.

ZonPV, 2016-01-01:
Is er informatie beschikbaar hoe er getest kan worden met simulator tijdens de uren dat de inverter niet actief is? Er wordt zoiets gesuggereerd met de directory Development. De settings hiervoor zijn me niet helder.

ZonPV, 2016-01-02:
Tegenvallende resultaten welke te zien waren in de logfile:
2016-01-02 15:04:22,534 DEBUG Importing output plugin PVoutputOutput
2016-01-02 15:04:22,675 INFO connecting to 192.168.1.22 port 8899
2016-01-02 15:04:23,746 INFO ID: NLDN1020138B1xxx
2016-01-02 15:04:23,747 DEBUG Run pluginPVoutputOutput

Maar soms ook:
2016-01-02 15:05:46,702 DEBUG Importing output plugin PVoutputOutput
2016-01-02 15:05:46,835 INFO connecting to 192.168.1.22 port 8899
2016-01-02 15:05:47,951 INFO ID: NLDN1020138B1xxx
2016-01-02 15:05:47,952 DEBUG Run pluginPVoutputOutput
2016-01-02 15:05:47,998 INFO Uploading to PVoutput
2016-01-02 15:05:48,261 INFO OK 200: Added Status

Beide waren het gevolg van handmatig starten via console.

Kan iemand mij uitleggen waarom deze handmatige actie steeds weer wisselende resultaten oplevert? 
Heb tussentijds geen aanpassingen in scripts of op PVoutput.org gedaan...
Ook blijkt Taakplanner geen succesvolle uploads geeft. Had verwacht dat die acties ook in de file omnik-export.log bijgeschreven?

Bswanny, 2016-01-02:
Hoi Pieter,

Volgens mij accepteert pvout slechts eens in de 5 minuten input. Hierdoor lijkt het dat het misgaat.

Met vriendelijke groet,

Bart

Woutrrr, 2016-01-02:
De pvoutput output werkt alleen op een vijf minuten interval (0, 5, 10, ..., 55). Dus dat zal de wisselende resultaten verklaren.

Dat het via de Taakplanner geen succesvolle uploads geeft, blijft vreemd. Kan je iets meer details geven over hoe je die hebt ingesteld, wat voor systeem je gebruikt e.d. ?

Woutrrr, 2016-01-02:
Het simulator script in Development leest waardes van zes uur geleden uit een mysql database en geeft deze terug. Om de simulator dus te kunnen gebruiken moet wel eerst de database gevuld worden door de MysqlOutput van het OmnikExport script.

ZonPV, 2016-01-02:
@Bswanny en Woutrrr : Is plausibel. Ik hoop het morgen te ontdekken als mijn taakplanner ook goed werkt. Helaas is hij momenteel niet aanspreekbaar...
@Woutrrr : Ik heb een Synology DS710+ met meest recente firmware draaien. In Taakplanner die het OS biedt heb ik een job aangemaakt. Daarbij ook het juiste pad ingesteld en de user ROOT gemachtigd om deze job elke 5 minuten uit te laten voeren tussen 8:00 en 18:00 uur.
Wat ik echter vanavond als mogelijke verklaring ontdekte is dat de file als extensie .py moet voeren in plaats van .sh
Ik geef het meteen toe: ben een beginner in Linux. Weet me een beetje te behelpen met WinSCP op zo'n apparaat.
Heb de file hernoemd in de job alsmede in het OS.
Dit is de inhoud van het bestand crontab.py:

*/5 * * * * /usr/bin/python /volume1/@appstore/Omnik/OmnikExport.py

Hieronder de opdrachtregel die ik in de taakplanner van de Synology heb staan:
/volume1/@appstore/Omnik/crontab.py

Heel fijn dat je in elk geval je applicatie belangeloos beschikbaar stelt en ons helpt!
Hopelijk kun je mijn laatste probleem ook nog oplossen? Bij voorbaat dank.
Pieter

ZonPV, 2016-01-03:
Zojuist getest met uploaden naar PVoutput:
Als ik op Syno handmatig vanuit Terminal het commando " python OmnikExport.py" geef op een vijfvoud van de minuut dan lees ik in logfile:
-----------------------------------------
2016-01-03 11:49:58,579 DEBUG Importing output plugin PVoutputOutput
2016-01-03 11:49:58,676 INFO connecting to 192.168.1.22 port 8899
2016-01-03 11:49:59,761 INFO ID: NLDN1020138B1009
2016-01-03 11:49:59,762 DEBUG Run pluginPVoutputOutput

2016-01-03 11:50:21,125 DEBUG Importing output plugin PVoutputOutput
2016-01-03 11:50:21,217 INFO connecting to 192.168.1.22 port 8899
2016-01-03 11:50:22,362 INFO ID: NLDN1020138B1009
2016-01-03 11:50:22,363 DEBUG Run pluginPVoutputOutput
2016-01-03 11:50:22,363 INFO Uploading to PVoutput
2016-01-03 11:50:22,613 INFO OK 200: Added Status

2016-01-03 11:55:07,031 DEBUG Importing output plugin PVoutputOutput
2016-01-03 11:55:07,122 INFO connecting to 192.168.1.22 port 8899
2016-01-03 11:55:08,279 INFO ID: NLDN1020138B1009
2016-01-03 11:55:08,279 DEBUG Run pluginPVoutputOutput
2016-01-03 11:55:08,280 INFO Uploading to PVoutput
2016-01-03 11:55:08,527 INFO OK 200: Added Status

2016-01-03 12:00:03,012 DEBUG Importing output plugin PVoutputOutput
2016-01-03 12:00:03,099 INFO connecting to 192.168.1.22 port 8899
2016-01-03 12:00:04,196 INFO ID: NLDN1020138B1009
2016-01-03 12:00:04,197 DEBUG Run pluginPVoutputOutput
2016-01-03 12:00:04,197 INFO Uploading to PVoutput
2016-01-03 12:00:06,299 INFO OK 200: Added Status
-------------------------------------------------
Heb daarbij het uitvoeren van het script via de taakplanner uitgezet omdat de taakplanner geen succesvolle uploads kon genereren.
Zou dat te maken kunnen hebben met wat Bswanny hierboven schreef:
-----------------
I know it has something to do with the location where the module is stored, but i don't know how to get this working.

For now i got it working by copying the PVoutputOutput.py into the Omnik-Data-Logger directory.
-----------------

Bswanny, 2016-01-03:
Pieter,

Waar het bij mij om ging was dat als je via cron de actie start,  hij dan niet het pad kon vinden naar de onderliggende bestandjes (zoals de PVoutputOutput.py). De actie start netjes en werd ook zonder foutmelding afgesloten, aangezien omnikexport gewoon zijn werk deed. Als je in de system logfiles (var/log/syslog) gaat kijken, dan zie je vaak wat meer commentaar. Hieruit heb ik destijds kunnen vinden waarom het misging. Door de output bestanden in dezelfde map te zetten als omnikexport.py, werd dit probleem bij mij verholpen. 

Bij mij draait alles op een raspberry, maar volgens mij maakt dat niet het verschil. 

Met vriendelijke groet,

Bart

ZonPV, 2016-01-03:
H Bart,
Bedankt voor je reactie. Helaas blijkt mijn Syno via de Taakplanner geen output te genereren..
Hierbij de inhoud van het script:

/volume1/@appstore/Omnik/crontab.py >> /volume1/@appstore/Omnik/cron.log

wordt wel uitgevoerd. Er wordt ook opgegeven file aangemaakt in opgegeven map als ik op direct uitvoeren klik. Maar deze file is leeg..., geen output. Weet niet wat ik daaruit moet afleiden.

Moet dus nog ff speuren.
Pieter

ZonPV, 2016-01-03:
H Wouter,
Zie dat je gisteren ook nog enkele aanpassingen hebt gedaan in o.a. configuratiefile. Zoals de optie: CSVoutput.
Is het mogelijk om meerdere plugins tegelijk uit te voeren? Weet de syntax daar niet voor. Geprobeerd met:
*enabled_plugins = PVoutputOutput*
*enabled_plugins = CSVOutput*

en ook door beiden op één regel te zetten. Maar wat is dan scheidingsteken? 
De omschrijving plugins suggereert meervoud..

En wat wordt dan de naam en lokatie van het outputbestand?? 

Bij Log-gedeelte geef je aan: *(combinations are possible)* 
Maar als ik de komma (,) als seperator gebruik geeft dat een error.

Met een CSV-file zou ik erg geholpen zijn!
Alvast bedankt voor reactie.

Met vriendelijke groet,
Pieter

ZonPV, 2016-01-08:
>>H Bart,
>> Bedankt voor je reactie. Helaas blijkt mijn Syno via de Taakplanner geen output te genereren..
>> Hierbij de inhoud van het script:

>> /volume1/@appstore/Omnik/crontab.py >> /volume1/@appstore/Omnik/cron.log

>> wordt wel uitgevoerd. Er wordt ook opgegeven file aangemaakt in opgegeven map als ik op direct >> uitvoeren klik. Maar deze file is leeg..., geen output. Weet niet wat ik daaruit moet afleiden.

>> Moet dus nog ff speuren.
>> Pieter

Opgelost! 
Op de Synology in Taakplanner geen file met script aanroepen maar direct de opdrachtregel ingeven: 

*python /volume1/@appstore/Omnik/OmnikExport.py*

We leren elke dag bij...

Nu nog uitzoeken hoe ik CSV-file kan genereren. Wie helpt me op weg?

Woutrrr, 2016-01-08:
De verschillende plugins kunnen geactiveerd worden door er een komma tussen te plaatsen. Dus 
```
enabled_plugins = CSVOutput,PVOutputOutput
```
Let op, de namen zijn hoofdlettergevoelig en ook mag er geen spatie tussen de outputs staan.

Bij de loggers heb je en bug gevonden, want dit zou hetzelfde moeten werken. Ik heb een nieuwe versie online gezet waarin meerdere loggers wel werken.

ZonPV, 2016-01-09:
Dag Wouter,
Bedankt voor reactie! Heb nieuwe versie gedownload en geïnstalleerd. Gaf in eerste instantie foutmelding

*De opdracht 'python OmnikExport.py' gaf foutcode 1 terug, samen met de foutmelding Traceback (most recent call last):
File "OmnikExport.py", line 170, in <module> omnik_exporter.run()
File "OmnikExport.py", line 52, in run
_import_(plugin_name)
ImportError: No module named PVOutputOutput.*

Ik heb toen in de map ..\outputs het bestand PVoutputOutput hernoemd waarna de melding niet meer terug kwam en (handmatig) uploaden naar de site weer werkte.

De logfile:

*2016-01-09 12:40:02,148 DEBUG Importing output plugin PVOutputOutput
2016-01-09 12:40:02,230 DEBUG Importing output plugin CSVOutput
2016-01-09 12:40:02,231 INFO connecting to 192.168.1.22 port 8899
2016-01-09 12:40:02,749 DEBUG Importing output plugin PVOutputOutput
2016-01-09 12:40:03,479 INFO ID: NLDN1020138B1009
2016-01-09 12:40:03,480 DEBUG Run pluginPVoutputOutput
2016-01-09 12:40:03,481 INFO Uploading to PVoutput
2016-01-09 12:40:03,733 INFO OK 200: Added Status
2016-01-09 12:40:03,733 DEBUG Run pluginCSVOutput*

Wat me echter opvalt is de verschillende (hoofd-)lettergebruik in namen 
*2016-01-09 12:40:03,480 DEBUG Run pluginPVoutputOutput*
en 
*2016-01-09 12:40:03,480 DEBUG Run pluginPVOutputOutput*

Zit daar misschien ook nog een foutje in? Want mijn Taakplanner werkt opeens niet meer. Ik kan nog wel handmatig uploaden naar PVoutput.com als ik om de vijf minuten handmatig de Omnik aanroep via 'python OmnikExport.py'
Heel jammer; automatisch uploaden naar PVoutput werkte feilloos...

Oplossing hiervoor:
Kopieren van het bestand: PVOutputOutput.py naar de hoofdmap waar ook OmnikExport.py staat...

Bovenstaande is voor mij dus geen echt probleem meer want ik heb een workaround.

Echter: Ik heb nog onduidelijkheid over uitvoer naar CSV-bestand. Zie nu wel in de commandline-interface de output van csv-output verschijnen:
*Id,Temp,VPV1,VPV2,VPV3,IPV1,IPV2,IPV3,IAC1,IAC2,IAC3,VAC1,VAC2,VAC3,FAC1,PAC1,FAC2,PAC2,FAC3,PAC3,ETODAY,ETOTAL,HTOTAL
NLDN1020138B1009,19.2,90.8,0.0,-1,0.4,18.1,-1,0.1,-1,-1,233.5,-1,-1,50.0,29,-1,-1,-1,-1,0.13,2.7,44*

Hoe krijg ik dat voor elkaar? Want in geen enkele map is een bestand met csv-opmaak of bovenstaande info te vinden.
Hierbij het naar mijn idee relevante gedeelte van de config.cfg:
*Type=file*

Ik dacht zo ook het probleem met CSVOutput mogelijk op te lossen door dat bestand ook naar hoofdmap te zetten. Maar dat was geen oplossing..
Heb nu de volgende output in de logfile:
*2016-01-09 13:55:02,927 DEBUG Importing output plugin PVOutputOutput
2016-01-09 13:55:03,010 DEBUG Importing output plugin CSVOutput
2016-01-09 13:55:03,011 INFO connecting to 192.168.1.22 port 8899
2016-01-09 13:55:04,147 INFO ID: NLDN1020138B1009
2016-01-09 13:55:04,148 DEBUG Run pluginPVoutputOutput
2016-01-09 13:55:04,149 INFO Uploading to PVoutput
2016-01-09 13:55:04,397 INFO OK 200: Added Status
2016-01-09 13:55:04,397 DEBUG Run pluginCSVOutput*

Mogelijk weet jij nog suggestie aan te dragen? In elk geval alvast dank voor verder meedenken!

Gr.
Pieter

Woutrrr, 2016-01-09:
Ah, PVOutputOutput was een foutje in het typen van mij. Dat had PVoutputOutput moeten zijn (zoals de bestandsnaam).

Ook heb ik gekeken naar waarom de plugins bij jou niet gevonden konden worden. Ik vermoed dat dat komt doordat de synology taakplanner het script met een andere working dir uitvoert. Het script is aangepast om een volledig path the gebruiken voor de outputs directory. Dus waarschijnlijk is het niet meer nodig om de bestanden uit die map te verplaatsen.

Over de CSV output; die is gemaakt voor iemand die het script wilde gebruiken in een ander programma. Hiervoor was het niet nodig om de csv data naar een bestand te schrijven. Om de csv output op de console naar een file te krijgen kan je '>> /path/naar/je/csv/file.csv' toevoegen aan de opdracht regel in de taakplanner. Wel is het dan handig als je de header van de csvoutput uitzet, anders krijg je die er telkens weer bij, ook moet de logging dan niet op console staan om dezelfde reden.

Ik hoop dat je hier weer mee verder kunt.

ZonPV, 2016-01-10:
Hoi Wouter,
You made my day! Ook uitvoeren naar CSV-file is nu opgelost! Was me al in aan het lezen op Linux tutuorials maar dat is voor mij geen feest.
Commandoregel in Taakplanner aangevuld. Tevens in config.cfg csv-file zonder header ingesteld:
*[csv]
disable_header = true*
Meetgegevens kunnen nu ingelezen worden op mijn webserver. Ik ben heel blij met je tool!
Nogmaals hartelijk dank!

hlubach, 2016-01-29:
Ook ik loop tegen het feit aan dat het log leeg blijft.
Ik had nog een oud script, gisteren dus de nieuwe geïnstalleerd  , en kreeg de foutmeldingen.
Dus lezen , heb de regel met de plugins aangepast met PVoutputOutput  en toe geen fouten meer.
Uiteraard de check gedaan met LiveStats en OmnikExport en toen bleek dat het log leeg bleef.
Verder lezend  stond er iets over het sysid en api key maar dat staat als normaal in het cfg.
Toen maar bij de Log Output achter console,file  geplaatst .
Nu is het log file niet leeg maar verschijnt dit als resultaat
2016-01-29 10:44:51,570 DEBUG Importing output plugin PVoutputOutput
2016-01-29 10:44:51,663 INFO connecting to 192.168.178.115 port 8899
2016-01-29 10:44:52,296 INFO ID: NLDN2020132S2067
2016-01-29 10:44:52,298 DEBUG Run pluginPVoutputOutput

Alle hulp is welkom

Met vriendelijke groet

ZonPV, 2016-01-29:
Kreeg exact hetzelfde verschijnsel: ook met juiste inloggegevens en api-key: geen foutmelding maar ook geen bevestiging.
Probeer eens op een veelvoud van 5 minuten (handmatig) je data naar PVoutput.org te verzenden.

*2016-01-28 16:10:02,403 DEBUG Importing output plugin PVOutputOutput
2016-01-28 16:10:02,485 DEBUG Importing output plugin CSVOutput
2016-01-28 16:10:02,486 INFO connecting to 192.168.1.22 port 8899
2016-01-28 16:10:03,640 INFO ID: NLDN1020138B1009
2016-01-28 16:10:03,640 DEBUG Run pluginPVoutputOutput
2016-01-28 16:10:03,641 INFO Uploading to PVoutput
2016-01-28 16:10:03,887 INFO OK 200: Added Status
2016-01-28 16:10:03,887 DEBUG Run pluginCSVOutput*

hlubach, 2016-01-29:
Schijnt nu te werken bij mij.
Krijg nu op de site gewoon de data te zien.
Enkel in de applicatie op de telefoon geen getallen maar wel de grafieken
en in de nieuwste applicatie wel de data

ottertje1982, 2016-09-05:
Ik moest mijn raspberry opnieuw inrichten.
Toen heb ik gelijk de nieuwste versie van Omnik-DataLogger geinstalleerd, maar ik loop tegen een probleem aan waar ik even de oplossing niet van zie.

Als ik python OmnikExport.py draai krijg ik de volgende foutmelding:

2016-09-05 22:13:49,209 DEBUG Importing output plugin
Traceback (most recent call last):
  File "OmnikExport.py", line 170, in <module>
    omnik_exporter.run()
  File "OmnikExport.py", line 52, in run
    __import__(plugin_name)
ValueError: Empty module name

Python LiveStat.py werkt wel, want dan krijg ik gewoon de data te zien vanuit mijn omvormer

JPBTEW, 2017-01-21:
Zijn er geen reacties gekomen op deze vraag?
 ik zit met hetzelfde probleem....... Hopelijk weet iemand de oplossing...

hlubach, 2017-02-03:
LibreELEC:~/omnik # python OmnikExport.py
Traceback (most recent call last):
  File "OmnikExport.py", line 170, in <module>
    omnik_exporter.run()
  File "OmnikExport.py", line 52, in run
    __import__(plugin_name)
ValueError: Empty module name

Door het opnieuw installeren van Libreelec de oude omnik gewist Nu de nieuwe erop maar deze foutmelding

hlubach, 2017-02-04:
in de 1e alinea staat dat je de module nog moet invullen 
PVoutputOutput

Ben enkel nog  de regel kwijt log  enabled  true or false

hlubach, 2017-06-26:
LibreELEC:~/Omnik # python OmnikExport.py
Traceback (most recent call last):
  File "OmnikExport.py", line 170, in <module>
    omnik_exporter.run()
  File "OmnikExport.py", line 38, in run
    self.build_logger(self.config)
  File "OmnikExport.py", line 114, in build_logger
    'level': log_levels[config.get('log', 'level')]
KeyError: 'info. debug'

After a new install this error  when trying to export the file.
Part config file
################
### Settings ###
################

[general]
# General:enabled_plugins
# Choose which outputs to use
# Possible options: MysqlOutput,PVoutputOutput,ConsoleOutput,CSVOutput
enabled_plugins = PVoutputOutput

Ca somebody help

hlubach, 2018-03-02:
Skip to content
This repository

    Pull requests
    Issues
    Marketplace
    Explore

    @hlubach

12
18

    26

Woutrrr/Omnik-Data-Logger
Code
Issues 11
Pull requests 1
Projects 0
Wiki
Insights
using 2 inverters with 1 logger #26
Open
refthoom opened this Issue on 21 Apr 2017 · 16 comments
Comments
Assignees
No one assigned
Labels
None yet
Projects
None yet
Milestone
No milestone
Notifications

You’re receiving notifications because you commented.
3 participants
@refthoom
@XtheOne
@hlubach
@refthoom
refthoom commented on 21 Apr 2017 •

Hi Wouterrr,

Installed your logger scripts and it worked out-of-the-box, kudos for you! :-) However...

I have a separate Wifi box (1 device with SN 504XXXXXX) that uses RS485 (in a daisy chain) to connect to my 2 inverters. So no individual Wifi-kits on the inverters themselves. Livestats.py produces a neat message as expected, but shows only values of the first inverter, the second one is not shown.

Assuming that the received string holds data of both inverters, is there a way to extract the second one as wel? I know both ID's of the inverters and this could be added to config.cfg so the script could look up the string and parse from there?

I'm no programmer but pretty handy and willing to help/test whatever. TIA

Edit:
I saw this: InverterMsg.py on line 8 says:
def init(self, msg, offset=0):
Could this offset be used to parse several parts of the response message? I tried offsetting here but no visible effect.

Edit2: Can anyone point to the part of the code where I need to iterate parsing the message for more inverters?
@XtheOne
XtheOne commented on 13 Jul 2017

Did you solve this?
You can look if both inverter data is in the returned data.
Add this to OmnikExport.py at line 90 (between "inverter_socket.close()" and "msg = InverterMsg.InverterMsg(data)")

self.logger.info('RAW Packet (len={0}): '.format(len(data))+':'.join(x.encode('hex') for x in data))

This will log a hex dump of the returned data.
[... 159 lines cut ...]
PV3 V: -1 I: -1
L1 P: 1319 V: 231.6 I: 5.6 F: 50.02
L2 P: -1 V: -1 I: -1 F: -1
L3 P: -1 V: -1 I: -1 F: -1
LibreELEC:~ #

OmniExport also:
LibreELEC:~ # python /storage/Omnik/OmnikExport.py
2018-03-02 12:48:49,747 DEBUG Importing output plugin PVoutputOutput
2018-03-02 12:48:49,878 INFO connecting to 192.168.178.115 port 8899
2018-03-02 12:48:50,506 INFO ID: NLDN2020132S2067
2018-03-02 12:48:50,508 DEBUG Run pluginPVoutputOutput
LibreELEC:~ #

But on pvoutput no change in data stiil 1-3-2018

This part is from the log file

2018-03-02 12:52:10,634 DEBUG Importing output plugin PVoutputOutput
2018-03-02 12:52:10,768 INFO connecting to 192.168.178.115 port 8899
2018-03-02 12:52:11,411 INFO ID: NLDN2020132S2067
2018-03-02 12:52:11,413 DEBUG Run pluginPVoutputOutput

can't find the error
@hlubach

refthoom, 2018-03-02:
@hlubach
I suggest you take a look at the fork from Xtheone and use that with a fresh install. Much improved script with more and better error notifications.

hlubach, 2018-03-02:
Did what you had suggested
with this as result

ibreELEC:~/Omnik # python LiveStats.py
2018-03-02 17:41:56,112 DEBUG Importing output plugin ConsoleOutput
2018-03-02 17:41:59,231 INFO Loggers found  on the network: ['192.168.178.115', '602538088'].
2018-03-02 17:41:59,232 INFO Connecting to logger with IP: 192.168.178.115 and SN 602538088
2018-03-02 17:41:59,233 INFO connecting to 192.168.178.115 port 8899
2018-03-02 17:41:59,246 DEBUG RAW sent Packet (len=16): 68:02:41:b1:68:00:ea:23:68:00:ea:23:01:00:df:16  hAh#h#
2018-03-02 17:41:59,887 DEBUG RAW received Packet (len=99): 68:55:41:b0:68:00:ea:23:68:00:ea:23:81:02:01:4e:4c:44:4e:32:30:32:30:31:33:32:53:32:30:36:37:00:ec:0a:ba:00:00:ff:ff:00:03:00:b5:ff:ff:00:03:ff:ff:ff:ff:08:d4:ff:ff:ff:ff:13:89:00:48:ff:ff:ff:ff:ff:ff:ff:ff:02:a2:00:01:4a:e4:00:00:4d:47:00:01:00:00:00:00:ff:ff:00:00:00:00:00:00:00:00:00:00:d9:16  hUAh#h#NLDN2020132S2067HJMG
2018-03-02 17:41:59,888 DEBUG DATA len=85: 
2018-03-02 17:41:59,890 INFO Inverter ID: NLDN2020132S2067
Traceback (most recent call last):
  File "LiveStats.py", line 17, in <module>
    inverter_exporter.run()
  File "/storage/Omnik/InverterExport.py", line 150, in run
    self.logger.info("Inverter main firmware version: {0}".format(msg.main_fwver))

refthoom, 2018-03-05:
Check svp of je de hele fork van Xtheone, dus alle files ook de
subdirectories, hebt gekopieerd naar je Rpi. Bekijk dan de config file want
die is compleet anders geworden.

De code is best solide, deze fout lijkt het meest op of restant van oude
files, of een oude/incomplete config file

hlubach, 2018-03-05:
Alles nog een keer gedownload  en uit gepakt
Nu niet op de RPI maar gewoon laptop

Dezelfde foutmelding

han@han-Satellite-L875 ~/Downloads/Inverter-Data-Logger-master $ python
LiveStats.py
2018-03-05 16:49:37,634 DEBUG Importing output plugin ConsoleOutput
2018-03-05 16:49:37,635 INFO Connecting to logger with IP: 192.168.178.115
and SN 602538088
2018-03-05 16:49:37,635 INFO connecting to 192.168.178.115 port 8899
2018-03-05 16:49:37,707 DEBUG RAW sent Packet (len=16):
68:02:41:b1:68:00:ea:23:68:00:ea:23:01:00:df:16  hAh#h#
2018-03-05 16:49:38,342 DEBUG RAW received Packet (len=99):
68:55:41:b0:68:00:ea:23:68:00:ea:23:81:02:01:4e:4c:44:4e:32:30:32:30:31:33:32:53:32:30:36:37:01:20:0a:ed:00:00:ff:ff:00:09:00:b6:ff:ff:00:09:ff:ff:ff:ff:09:0c:ff:ff:ff:ff:13:8c:00:e8:ff:ff:ff:ff:ff:ff:ff:ff:02:55:00:01:4b:77:00:00:4d:68:00:01:00:00:00:00:ff:ff:00:00:00:00:00:00:00:00:00:00:92:16
hUAh#h#NLDN2020132S2067 UKwMh
2018-03-05 16:49:38,342 DEBUG DATA len=85:
2018-03-05 16:49:38,342 INFO Inverter ID: NLDN2020132S2067
Traceback (most recent call last):
  File "LiveStats.py", line 17, in <module>
    inverter_exporter.run()
  File "/home/han/Downloads/Inverter-Data-Logger-master/InverterExport.py",
line 150, in run
    self.logger.info("Inverter main firmware version:
{0}".format(msg.main_fwver))
  File "/home/han/Downloads/Inverter-Data-Logger-master/InverterMsg.py",
line 154, in main_fwver
    if (self.__get_int(101) == 0): return ""
  File "/home/han/Downloads/Inverter-Data-Logger-master/InverterMsg.py",
line 33, in __get_int
    return int(self.raw_msg[begin:begin+1].encode('hex'), 16)
ValueError: invalid literal for int() with base 16: ''

Kan r niet achter komen wat niet goed zit

refthoom, 2018-03-05:
Ik zie het ook niet direct. Heb je ook de config vanaf scratch ingevuld?
Heb je de auto discovery gebruikt of heb je zelf je ip adres en logger id
opgegeven?
Kan er ergens een non-printable character of een spatie uit een copy-paste
actie in de config zijn gekomen?

Kom je hier niet mee verder dan weet ik het ook niet, ben geen python
expert. Sorry.

> Check svp of je de hele fork van Xtheone, dus alle files ook de
> subdirectories, hebt gekopieerd naar je Rpi. Bekijk dan de config file
want
> die is compleet anders geworden.
>
> De code is best solide, deze fout lijkt het meest op of restant van oude
> files, of een oude/incomplete config file
>
>
>
>
>
>
> Did what you had suggested
> with this as result
>
> ibreELEC:~/Omnik # python LiveStats.py
> 2018-03-02 17:41:56,112 DEBUG Importing output plugin ConsoleOutput
> 2018-03-02 17:41:59,231 INFO Loggers found on the network:
> ['192.168.178.115', '602538088'].
> 2018-03-02 17:41:59,232 INFO Connecting to logger with IP: 192.168.178.115
> and SN 602538088
> 2018-03-02 17:41:59,233 INFO connecting to 192.168.178.115 port 8899
> 2018-03-02 17:41:59,246 DEBUG RAW sent Packet (len=16):
> 68:02:41:b1:68:00:ea:23:68:00:ea:23:01:00:df:16 hAh#h#
> 2018-03-02 17:41:59,887 DEBUG RAW received Packet (len=99):
> 68:55:41:b0:68:00:ea:23:68:00:ea:23:81:02:01:4e:4c:44:4e:32:
> 30:32:30:31:33:32:53:32:30:36:37:00:ec:0a:ba:00:00:ff:ff:00:
> 03:00:b5:ff:ff:00:03:ff:ff:ff:ff:08:d4:ff:ff:ff:ff:13:89:00:
> 48:ff:ff:ff:ff:ff:ff:ff:ff:02:a2:00:01:4a:e4:00:00:4d:47:00:
> 01:00:00:00:00:ff:ff:00:00:00:00:00:00:00:00:00:00:d9:16
> hUAh#h#NLDN2020132S2067HJMG
> 2018-03-02 17:41:59,888 DEBUG DATA len=85:
> 2018-03-02 17:41:59,890 INFO Inverter ID: NLDN2020132S2067
>
> Traceback (most recent call last):
> File "LiveStats.py", line 17, in
> inverter_exporter.run()
> File "/storage/Omnik/InverterExport.py", line 150, in run
> self.logger.info("Inverter main firmware version:
> {0}".format(msg.main_fwver))
>
>
> <https://github.com/Woutrrr/Omnik-Data-Logger/issues/21#
> issuecomment-369977848>,
> AakE19trVpnHJFdDBbCHYUd2Ed4Wx5dvks5taXaogaJpZM4G6kbT>

> <https://github.com/Woutrrr/Omnik-Data-Logger/issues/21#
issuecomment-370451964>,
7nTXnoFS7rPGkBtjVerL2iBtcks5tbVcbgaJpZM4G6kbT>

hlubach, 2018-03-05:
Vanaf scratch, eigen IP ingevuld etc

hlubach, 2018-03-14:
Nog  maar en keer  de Data logger opnieuw geinstallerd op zowel op de laptop als de Rpi .
Op de laptop draait het nu  

han@han-Satellite-L875 ~/Thelma Houston - $ python OmnikExport.py
2018-03-14 16:25:59,056 DEBUG Importing output plugin PVoutputOutput
2018-03-14 16:25:59,069 INFO connecting to 192.168.178.115 port 8899
2018-03-14 16:25:59,719 INFO ID: NLDN2020132S2067
2018-03-14 16:25:59,719 DEBUG Run pluginPVoutputOutput
2018-03-14 16:25:59,719 INFO Uploading to PVoutput
2018-03-14 16:26:00,015 INFO OK 200: Added Status

maar op de Rpi wordt de added status niet vermeld na de opdracht
python OmnikExport.py

LibreELEC:~ # python  /storage/omnik/OmnikExport.py
2018-03-14 16:39:43,155 DEBUG Importing output plugin PVoutputOutput
2018-03-14 16:39:43,298 INFO connecting to 192.168.178.115 port 8899
2018-03-14 16:39:43,994 INFO ID: NLDN2020132S2067
2018-03-14 16:39:43,996 DEBUG Run pluginPVoutputOutput

Wie weet wat er aan de hand kan zijn

refthoom, 2018-03-16:
Als het op je laptop wel werkt bewijst dit dat het script in orde is. Er
blijft dan niet zoveel over waar het aan kan liggen;

Heb je de installatie op de rpi gedaan vanuit een *nieuwe* directory met
"git pull https://github.com/XtheOne/Inverter-Data-Logger.git"? Dan weet je
zeker dat je geen oude bestanden hebt die in de weg zitten.

Als dat ook geen soelaas biedt kan het alleen nog aan de python
versie/installatie/configuratie liggen maar daar weet ik echt niets van.

XtheOne, 2018-03-19:
Er wordt alleen op de hele 5 minuten DATA gestuurd.
2018-03-14 16:39:43,996 DEBUG Run pluginPVoutputOutput
Niks sturen.

2018-03-14 16:25:59,719 DEBUG Run pluginPVoutputOutput
een hele 5 minuten moment, stuurt DATA.

De code die dit regelt:
if (now.minute % 5) == 0:  # Only run at every 5 minute interval

XtheOne, 2018-03-19:
Als het om mijn code gaat graag daar een issue openen, hier kijk ik vrijwel niet meer.

hlubach, 2019-11-21:
Moest de RPI opnieuw installeren , dus ook de Omnik eraf.
nu staat de Omnik erop maar schijnt er geen gegevens naar pvoutput te sturen 

In de terminal  doe ik onderstaande .
python ./Omnik/Livestats.py met resultaat
python ./Omnik/OmnikExport.py geeft het volgende
LibreELEC:~ # python ./Omnik/LiveStats.py
2019-11-21 13:22:22,714 DEBUG Importing output plugin ConsoleOutput
2019-11-21 13:22:22,717 INFO connecting to 192.168.178.115 port 8899
2019-11-21 13:22:23,356 INFO ID: NLDN2020132S2067
2019-11-21 13:22:23,358 DEBUG Run pluginConsoleOutput
ID: NLDN2020132S2067
E Today:  0.41   Total: 12023.0
H Total: 27637   Temp:  24.1 
PV1   V: 282.7   I:  0.8
PV2   V:   0.0   I: 18.1
PV3   V:    -1   I:   -1
L1    P:   190   V: 230.8   I:  0.8   F: 50.03
L2    P:    -1   V:    -1   I:   -1   F:    -1
L3    P:    -1   V:    -1   I:   -1   F:    -1
LibreELEC:~ # 
LibreELEC:~ # python ./Omnik/OmnikExport.py
2019-11-21 13:22:29,167 DEBUG Importing output plugin PVoutputOutput
2019-11-21 13:22:29,310 INFO connecting to 192.168.178.115 port 8899
2019-11-21 13:22:29,939 INFO ID: NLDN2020132S2067
2019-11-21 13:22:29,941 DEBUG Run pluginPVoutputOutput
LibreELEC:~ # 

Zo te zien gaan er dus data naar PV output 

de settings 
################
### Settings ###
################

[general]
# General:enabled_plugins
# Choose which outputs to use
# Possible options: MysqlOutput,PVoutputOutput,ConsoleOutput,CSVOutput
enabled_plugins = PVoutputOutput

Wie brengrt er licht in de tunnel
met vriendelijke groet

Woutrrr, 2019-11-21:
De PVoutput stuurt alleen data naar PVoutput op de 5 hele minuten (dus om 0, 5, 10, 15, etc).
Je lijkt het script uit te voeren om 22min, dus dat doet het script niks.

> 2019-11-21 13:22:29,941 DEBUG Run pluginPVoutputOutput

Dit wordt geregeld in "PVoutputOutput.py" op regel 19 ([hier](https://github.com/Woutrrr/Omnik-Data-Logger/blob/526085c25cfe2d9ed216f0094b6e01ab54c5a175/outputs/PVoutputOutput.py#L19))
`if (now.minute % 5) == 0:  # Only run at every 5 minute interval`

Dus als je het script automatisch laat uitvoeren op een interval dat op de 5 hele minuten loopt, zal het waarschijnlijk wel werken.

ottertje1982, 2019-11-21:
Hoe heb je de regel in je cronjob staan ?
Of heb je die nog niet ingesteld ?

Mocht je cronjob ingesteld hebben wat staat er in je logging in de map van Omnik

Info zoals ik dit zie bij mij:

2019-11-21 16:05:03,606 INFO ID: NLBN202015752154
2019-11-21 16:05:03,801 INFO Uploading to PVoutput
2019-11-21 16:05:04,146 INFO OK 200: Added Status

hlubach, 2019-11-21:
Als ik het dus goed begrijp
als ik vanaf de terminal het commando python ./Omnik/LiveStart.py doe , dan wordt er niets naar PV out gestuurd.
Pas als ik dit commando geen op 5 of 10 etc

crontab
0/5 7-23 * * * python ./Omnik/OmnikExport.py

ottertje1982, 2019-11-21:
Ik heb het in mijn crontab als volgt

*/5 * * * * /usr/bin/python

Weet niet of 0/5 of */5 verschil maakt

hlubach, 2019-11-21:
# python ./Omnik/OmnikExport.py
2019-11-21 16:30:38,838 DEBUG Importing output plugin PVoutputOutput
2019-11-21 16:30:38,983 INFO connecting to 192.168.178.115 port 8899
2019-11-21 16:30:39,611 INFO ID: NLDN2020132S2067
2019-11-21 16:30:39,613 DEBUG Run pluginPVoutputOutput
2019-11-21 16:30:39,614 INFO Uploading to PVoutput
Traceback (most recent call last):
  File "./Omnik/OmnikExport.py", line 170, in <module>
    omnik_exporter.run()
  File "./Omnik/OmnikExport.py", line 82, in run
    plugin.process_message(msg)
  File "/storage/Omnik/outputs/PVoutputOutput.py", line 49, in process_message
    response = urllib2.urlopen(request_object)
  File "/usr/lib/python2.7/urllib2.py", line 154, in urlopen
  File "/usr/lib/python2.7/urllib2.py", line 435, in open
  File "/usr/lib/python2.7/urllib2.py", line 548, in http_response
  File "/usr/lib/python2.7/urllib2.py", line 473, in error
  File "/usr/lib/python2.7/urllib2.py", line 407, in _call_chain
  File "/usr/lib/python2.7/urllib2.py", line 556, in http_error_default
urllib2.HTTPError: HTTP Error 403: Forbidden

ottertje1982, 2019-11-21:
config file goed ingesteld omdat je een Error 403 krijgt ?

hlubach, 2019-11-21:
is de oude die ik gebruikte op de rpi voorfdat ik opnieuw moest beginnen

[general]
# General:enabled_plugins
# Choose which outputs to use
# Possible options: MysqlOutput,PVoutputOutput,ConsoleOutput,CSVOutput
enabled_plugins = PVoutputOutput

[inverter]
# IP address of your Omnik inverter
ip = *******************
# Default for a Omnik with Wifi module
port = 8899
# S/N of the wifi kit
wifi_sn = **************
#use temperature of inverter for pvoutput
use_temperature = true

[mysql]
# Host where the mysql server is active
host = 127.0.0.1
user =
pass =
database =

[pvout]
# These two can be found at http://pvoutput.org/account.jsp
apikey = *******
sysid  = ********

[csv]
disable_header = true

[log]
# Log:Output
# Possible options: none,console,file (combinations are possible)
# Use none to disable logging
type = console,file

# Log:level
# Possible options: critical, error, warning, info, debug
level = debug

# Log:filename
# Output file for file logger
filename = omnik-export.log

ottertje1982, 2019-11-21:
Weet niet of het veel uitmaakt en of mijn omnik anders is dan die van jouw, maar zolang het werkt, werkt het ;-)

Ik heb mijn config.cfg toegevoegd je kan eens kijken of die het doet.

file uiteraard even hernoemen naar config.cfg

[omnik_config.txt](https://github.com/Woutrrr/Omnik-Data-Logger/files/3875539/omnik_config.txt)

hlubach, 2019-11-23:
LibreELEC:~ # python ./Omnik/OmnikExport.py
2019-11-23 10:55:11,647 DEBUG Importing output plugin PVoutputOutput
2019-11-23 10:55:11,788 INFO connecting to 192.168.178.115 port 8899
2019-11-23 10:55:12,605 INFO ID: NLDN2020132S2067
2019-11-23 10:55:12,607 DEBUG Run pluginPVoutputOutput
2019-11-23 10:55:12,609 INFO Uploading to PVoutput
Traceback (most recent call last):
  File "./Omnik/OmnikExport.py", line 170, in <module>
    omnik_exporter.run()
  File "./Omnik/OmnikExport.py", line 82, in run
    plugin.process_message(msg)
  File "/storage/Omnik/outputs/PVoutputOutput.py", line 49, in process_message
    response = urllib2.urlopen(request_object)
  File "/usr/lib/python2.7/urllib2.py", line 154, in urlopen
  File "/usr/lib/python2.7/urllib2.py", line 435, in open
  File "/usr/lib/python2.7/urllib2.py", line 548, in http_response
  File "/usr/lib/python2.7/urllib2.py", line 473, in error
  File "/usr/lib/python2.7/urllib2.py", line 407, in _call_chain
  File "/usr/lib/python2.7/urllib2.py", line 556, in http_error_default
urllib2.HTTPError: HTTP Error 403: Forbidden
LibreELEC:~ # 

tegen deze error loop ik nu dus aan 
terwijl  de Omnik data logger een copie was  ,van die ik altijd gebruikte 
Wat is er mis

ottertje1982, 2019-11-23:
Kan je google of een andere website pingen vanaf je machine ?

dus dan wel via ping www.google.com etc.

hlubach, 2019-11-25:
De melding komt van beide bronnen 
Van de raspberrie  LibreELEC:~ # 
 maar ook van de laptop han@han-HP-EliteBook-8770w

hlubach, 2019-11-27:
Heb na zeven dagen de fout gevonden 
Zit hem in de api key  daar had ik de korte string van  en daarom werkte het dus niet 
Dus het werkt weer
Met dank aan ieder

ottertje1982, 2019-11-28:
> Heb na zeven dagen de fout gevonden
> Zit hem in de api key daar had ik de korte string van en daarom werkte het dus niet
> Dus het werkt weer
> Met dank aan ieder

Ik zag dat je in je crontab het volgende had staan.

`0/5 7-23 * * * python ./Omnik/OmnikExport.py`

Wat je ook eventueel kunt doen is het volgende aanpassen in je OmnikExport.py

`if pvout_enabled and msg.getTemp() < 1000 and (now.minute % 5) == 0:
    if log_enabled:
        logger.info('Uploading to PVoutput')
    url = "http://pvoutput.org/service/r2/addstatus.jsp"`

Je zet hier alleen msg.getTemp() < 1000 tussen zodat de omvormer begint te loggen zodra deze aangaat.

In de zomer hoef je dan je Crontab niet te wijzigen

eterpstra, 2022-10-05:
Hij is oud maar ook ik loop tegen het volgende aan:
Uitlezen gaat prima.. Echter uploaden naar PVOUtput gaat niet of nauwelijks. 

Wel:

pi@raspberrypi:~/Omnik-Data-Logger $ python OmnikExport.py
2022-10-05 12:35:37,811 DEBUG Importing output plugin PVoutputOutput
2022-10-05 12:35:37,871 INFO connecting to 192.168.1.125 port 8899
2022-10-05 12:35:38,332 INFO ID: NLDN4020133S1052
2022-10-05 12:35:38,333 DEBUG Run pluginPVoutputOutput
2022-10-05 12:35:38,333 INFO Uploading to PVoutput
2022-10-05 12:35:38,616 INFO OK 200: Added Status

niet:
pi@raspberrypi:~/Omnik-Data-Logger $ python OmnikExport.py
2022-10-05 12:47:47,721 DEBUG Importing output plugin PVoutputOutput
2022-10-05 12:47:47,782 INFO connecting to 192.168.1.125 port 8899
2022-10-05 12:47:48,245 INFO ID: NLDN4020133S1052
2022-10-05 12:47:48,245 DEBUG Run pluginPVoutputOutput

crontab:
*/5 * * * * /usr/bin/python /home/pi/Omnik-Data-Logger/OmnikExport.py

Ik heb soms wel een uur gewacht en dan upload hij nog steeds niet.  Ook blijft de logfile leeg en kan ik dus niet zien wat er aan de hand is.

Het betreft een schone / nieuwe installatie van het script. Heeft iemand een oplossing?