Woutrrr/Omnik-Data-Logger #23: No data in Domoticz

noppie, 2016-02-22:
Hi Woutrrr,

I followed your instructions but I can't manage to get any data in Domoticz.
The logfiles are also empty, while the config file says logging enabled.
Used the correct IDX for the virtual sensors in Domoticz, but no data...
Any suggestions?

ikookmaar, 2016-03-18:
I'd like to have these instructions too, please.
The additions for exporting data to Domoticz described in https://www.domoticz.com/wiki/Omnik_Solar_Inverter won't work anymore, as config.cfg and OmnikExport.py have changed a lot since that version (september 2015), mainly through the use of "plugins". Maybe it's time for a Domoticz plugin?

motor4all, 2017-02-18:
Is the problem the config.cfg? Some syntaxes have change since before. If you want I can post an example

hans4307, 2017-02-20:
I've edited the Domoticz wiki. 
Getting your omnik data in Domoticz should be a piece of cake again.

noppie, 2017-02-20:
Thank you.
Followed your instructions but now i’m getting the next error:

File "DomoticzOutput.py", line 61
SyntaxError: Non-ASCII character '\xc2' in file DomoticzOutput.py on line 61, but no encoding declared; see http://www.python.org/peps/pep-0263.html for details
root@raspberrypi:/home/pi/Omnik-Data-Logger/Omnik-Data-Logger# python DomoticzOutput.py 

Any idea?

hans4307, 2017-02-20:
In the DomoticzOutput.py file, delete the degrees symbol in the comment at
line 61.
Line 61 is where the inverter temperature is send to domoticz.
I didn't test it yet, but hopefully this is the solution for your problem.
Please let me know if the problem isn't solved.

Hans.

noppie, 2017-02-20:
Yes I already figured that out, now i’m looking for a virtual sensor for my solar panles, but I don’t know wich one

hans4307, 2017-02-20:
I'm using the virtual sensor 'Electric (Instant + Counter)'.
This one is standard for recording the amount of electricity used, but when you click on the edit-button of the sensor in the utility tab, you can change the type from usage to return.

noppie, 2017-02-20:
Thanks,
How to change the icon….. now it looks like this:

noppie, 2017-02-21:
> Op 20 feb. 2017, om 22:42 heeft Norbert Otten <norbert.otten@gmail.com> het volgende geschreven:

Found it!  Just change from consumption to delivery…..

Everything works now!