Woutrrr/Omnik-Data-Logger #26: using 2 inverters

refthoom, 2017-04-21:
Hi Wouterrr,

Installed your logger scripts and it worked out-of-the-box, kudos for you! :-) However...

I have a separate Wifi box (1 device with SN 504XXXXXX) that uses RS485 (in a daisy chain) to connect to my 2 inverters. So no individual Wifi-kits on the inverters themselves. Livestats.py produces a neat message as expected, but shows only values of the first inverter, the second one is not shown. 

Assuming that the received string holds data of both inverters, is there a way to extract the second one as wel? I know both ID's of the inverters and this could be added to config.cfg so the script could look up the string and parse from there?

I'm no programmer but pretty handy and willing to help/test whatever. TIA

XtheOne, 2017-07-13:
Did you solve this?
You can look if both inverter data is in the returned data.
Add this to OmnikExport.py at line 90 (between "inverter_socket.close()" and "msg = InverterMsg.InverterMsg(data)")

self.logger.info('RAW Packet (len={0}): '.format(len(data))+':'.join(x.encode('hex') for x in data))

This will log a hex dump of the returned data.
On my 400TL2 I get 183 bytes in return.

refthoom, 2017-07-30:
Thx XtheOne for responding. No, not yet solved. I tried your suggestion, this is what it shows when running livestats.py:
RAW Packet (len=143): 68:81:51:b0:de:3d:10:1e:de:3d:10:1e:81:02:01:4e:4c:44:4e:32:30:32:30:31:33:38:39:34:30:32:33:01:79:08:fb:00:00:ff:ff:00:5e:00:b5:ff:ff:00:4f:ff:ff:ff:ff:09:1b:ff:ff:ff:ff:13:90:07:4a:ff:ff:ff:ff:ff:ff:ff:ff:00:b3:00:00:d4:79:00:00:2c:51:00:01:00:00:00:00:ff:ff:00:00:00:00:00:00:00:00:00:00:00:00:00:00:4e:4c:31:2d:56:31:2e:30:2d:30:30:34:33:2d:34:00:00:00:00:00:56:31:2e:36:2d:30:30:31:38:00:00:00:00:00:00:00:00:00:00:00:98:16
I don't know how to interpret this, maybe you do? 
TIA & regards

XtheOne, 2017-07-31:
This is data from 1 inverter.
Maybe there is an inverter offset number in the magic packet.
You can play with the values in OmnikExport.py lines 173 and 181

My guess would be to change this line:
response += ''.join(hex_list) + '\x01\x00' + checksum + '\x16'
to
response += ''.join(hex_list) + '\x02\x00' + checksum + '\x16'
And see if you then get data of the 2nd inverter

If the (new) Android app shows you data from both inverters you can use a packet sniffer to see how the app queries both inverters.

refthoom, 2017-07-31:
Sorry, that doesn't work. Following your suggestion it just waits a long time and then comes with a time out error:
Traceback (most recent call last):
  File "LiveStats.py", line 17, in <module>
    omnik_exporter.run()
  File "/home/fons/Omnik-Data-Logger/OmnikExport.py", line 73, in run
    data = inverter_socket.recv(1024)
socket.timeout: timed out

What I understand from the code it sends a magic packet that triggers the logger to send the data from the PV converters. The data received is then parsed like a string with the various bits in fixed positions. Since I have 1 logger for 2 PV converters the logger probably just sends a longer 'string' with the data of both in it, Or, that is my guess.

XtheOne, 2017-08-01:
The data dump contains values from 1 inverter and some version information.
Nothing that points to 2nd inverter data.
The trick is to find how to request the data from the 2nd inverter.
Either with the android app and packet sniffing, or disassembling the android or windows app.
Else ask Omnik for documentation.
I did look in a WiFi firmware update file but nothing found.
How are you now able to see data from both inverters?

refthoom, 2017-08-02:
So you're sure it's not the python script that 'chops' the end off the dump, making it look like it's data from only one inverter?
I haven't used the new app yet, will try that in a minute. However the website www.omnikportal.com shows data from both inverters.
Through the logger's interface I can add another server, where I guess it will send the data to. There is a test button to see if an added server is pingable or not. When I fill in my pi's IP the logger's test says ok. I can also change  the port but there is no reason so I'd better leave that on the default 8899. So I presume the data is now sent not only to the Omnik site, but also to the pi. 
So, how do I capture or read that data coming in on the pi?

XtheOne, 2017-08-03:
I am sure because the receive buffer is >1000 chars long and the dumped data contains the end byte (16)
If you set a server then the WiFi interface sends it's data to a web server running a listener.
It gets hard to test this without access to a 2 inverter setup.

Maybe if you sent the magic packet a 2nd time in the same session it returns the 2nd inverter data.

XtheOne, 2017-08-03:
I have just looked in the decoded source of the Android app and found:
concatStringToSend(Ljava/util/ArrayList;)Ljava/lang/String;
in com\omnik\omniksolar\android\ui\KitMainActivity.smali
It looks like more data from a array list is put in besides the serial number to select the inverter.

refthoom, 2017-08-04:
Really appreciate your time for this issue :-)

Isn't the app using data from their website, not the raw data from the loggerbox I mean?

The pi that the loggerbox is now configured to send data to, also runs iptables (because of pivpn). Is there an iptables command that you know of that I could use to show the data that it sends to the pi? I tried some commands but that either gives an error or just nothing at all (that a can find). 

Or if you give me an ip address of your own that I can point the logger to?

refthoom, 2017-08-04:
BTW this is what I get with: sudo tcpdump -v port 8899

tcpdump: listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
19:01:33.214844 IP (tos 0x0, ttl 64, id 1120, offset 0, flags [none], proto TCP (6), length 60)
    192.168.1.29.44838 > RaspiBase.8899: Flags [S], cksum 0xbca5 (correct), seq 59568119, win 16384, options [mss 1460,nop,wscale 0,nop,nop,TS val 1396 ecr 0], length 0
19:01:33.215146 IP (tos 0x0, ttl 64, id 49574, offset 0, flags [DF], proto TCP (6), length 40)
    RaspiBase.8899 > 192.168.1.29.44838: Flags [R.], cksum 0x66e1 (correct), seq 0, ack 59568120, win 0, length 0
19:01:34.213112 IP (tos 0x0, ttl 64, id 1123, offset 0, flags [none], proto TCP (6), length 60)
    192.168.1.29.20460 > RaspiBase.8899: Flags [S], cksum 0x5a9d (correct), seq 59814196, win 16384, options [mss 1460,nop,wscale 0,nop,nop,TS val 1398 ecr 0], length 0
19:01:34.213426 IP (tos 0x0, ttl 64, id 49580, offset 0, flags [DF], proto TCP (6), length 40)
    RaspiBase.8899 > 192.168.1.29.20460: Flags [R.], cksum 0x04db (correct), seq 0, ack 59814197, win 0, length 0
19:01:35.219965 IP (tos 0x0, ttl 64, id 1126, offset 0, flags [none], proto TCP (6), length 60)
    192.168.1.29.35799 > RaspiBase.8899: Flags [S], cksum 0x9a73 (correct), seq 60044653, win 16384, options [mss 1460,nop,wscale 0,nop,nop,TS val 1400 ecr 0], length 0
19:01:35.220273 IP (tos 0x0, ttl 64, id 49615, offset 0, flags [DF], proto TCP (6), length 40)
    RaspiBase.8899 > 192.168.1.29.35799: Flags [R.], cksum 0x44b3 (correct), seq 0, ack 60044654, win 0, length 0

XtheOne, 2017-08-05:
With the app you can connect locally, select manual mode (no login) at the login screen.
Hmm, the TCP dump does not show data.
You can also try the "solarman" app, in local mode they are the manufacturer of the WiFi boxes.
I am going on hollyday, for the next 3 weeks...
Maybe better that we continue by email later.

refthoom, 2017-08-23:
Hope your holiday was fine :-) I've been busy while you were away, with no real results unfortunately.

I tried both apps, Omnik's and Solarman's and the converters shwo up nicely but they give no clue on what is received (or sent). The Solarman app finds the logger first, then shows the converter serials and I have to select either one to retreive any data. This does look like it is possible to retreive data by providing a serial number but ofcourse I have no clue how to do that.

I sent a message to both Omnik and Solarman support for more technical documentation, but no response as of yet.

With tcpdump (sudo tcpdump -vv -x -X -U -s 65535 -i wlan0 port 8899 -w solarman.pcap) I can capture packets and its contents in a pcap file and open those with wireshark. It seems fine but I have little clue what I am looking at, to be honest. There is an incoming packet and an ack returned.

I could send the file to you, maybe you can see more?

An acquaintance of mine has some Python skills, I could ask him if he can dissect wouterrr's script and find a way to get more data by sending the magic packet.

XtheOne, 2017-08-30:
I am back, had a nice vacation! :-)
I did the same with tcpdump on my android phone and recorded the solarman app activities yesterday.
Will look into the data which is sent to the inverter, it seems that it is almost the same as the logger works, only the magic is different.
Can you mail me the pcap file?

refthoom, 2017-08-30:
I'd be happy to send an email!

I found two here I'll be trying the one that seems like the private email. If no answer in a few days we'll have to try something else :-)

XtheOne, 2017-09-07:
I got this working!
This can be found, together with other improvements, here:
https://github.com/XtheOne/Inverter-Data-Logger.git

hlubach, 2018-03-02:
Problem .
I have to make a fresh start on the RPI , therefore I had to install  the datalogger again.
Local I had still all the file , but after had copied them to the RPI , I can make contact to Pvoutput .
LiveStats works

LibreELEC:~ # python /storage/Omnik/LiveStats.py
2018-03-02 12:46:57,125 DEBUG Importing output plugin ConsoleOutput
2018-03-02 12:46:57,127 INFO connecting to 192.168.178.115 port 8899
2018-03-02 12:46:57,757 INFO ID: NLDN2020132S2067
2018-03-02 12:46:57,758 DEBUG Run pluginConsoleOutput
ID: NLDN2020132S2067
E Today:  2.39   Total: 8466.5
H Total: 19779   Temp:  28.4 
PV1   V: 280.0   I:  5.6
PV2   V:   0.0   I: 18.2
PV3   V:    -1   I:   -1
L1    P:  1319   V: 231.6   I:  5.6   F: 50.02
L2    P:    -1   V:    -1   I:   -1   F:    -1
L3    P:    -1   V:    -1   I:   -1   F:    -1
LibreELEC:~ # 

OmniExport also: