Woutrrr/Omnik-Data-Logger #27: Omnik ethernetcard support

abc1000, 2017-09-27:
Log via wifi works perfect.
But how to via ethernet? (Omnik module)

ldeng2, 2017-09-27:
Same problem for me. Somenone???

XtheOne, 2017-09-27:
Not much info is available of the new eternet interface logger protocol.
You can try the solarman android app and see if it finds the logger in local mode.
If yes my newer inverter logger scripts might work.

abc1000, 2017-09-27:
It's very new..
I tried the apps but useless..
These apps tried all wifi...

Can i connect manual to port omnik:8899? and see what's happening?

btw, for now livestats.py freezes my convertor..and must reset

XtheOne, 2017-09-27:
You can try my ScanLoggers.py script. If it does not list the inverter then it works differently then the WiFi logger which is based on iGEN firmware which uses protocol V4.
I cannot find any documentation on this ethernet interface.
How does the web interface looks like?

abc1000, 2017-09-27:
Sure, where can i find it?
Firmware is not changed.
webinterface has no wifi network but ethernet settings.
sn 190#

XtheOne, 2017-09-27:
https://github.com/XtheOne/Inverter-Data-Logger
Screenshot of the web interface?

abc1000, 2017-09-27:
Yes found it..

This script will look for iGEN WiFi Kit loggers from SolarMAN PV
List their IPs and S/Ns and connected inverters S/N
These loggers are found in Omnik, Hosola, Ginlong, Kstar, Seasun, SolaX, Samil, Sofar, Trannergy
and other Solar inverters
WiFi kit logger found, IP = 192.168.1.211 and S/N = 190#######
Listing Inverter(s) connected to this WiFi logger
closing socket, scanning done!
Traceback (most recent call last):
  File "ScanLoggers.py", line 79, in <module>
    data = logger_socket.recv(1500)
socket.timeout: timed out

[image]

XtheOne, 2017-09-27:
The good thing is that it looks like it is iGEN logger based.
It does connect on port 8899 but no data is received.
It could be that this logger firmware does not support V4 protocol anymore.
Did you try https://play.google.com/store/apps/details?id=com.yingzhen
Local mode, this app supports V5 protocol.
If it does not find the inverter then omnik or igen should be asked.

Maybe https://play.google.com/store/apps/details?id=com.igentech.config.activity
can find it?

abc1000, 2017-09-27:
Tried both apps allready but they  searching for AP's..

XtheOne, 2017-09-27:
Ok, then I will ask around tomorrow

abc1000, 2017-09-27:
ok, tnx

XtheOne, 2017-09-28:
I have sent a request for more information to the Omnik dev team.
I'm not related to Omnik myself but I use their inverters.

It does reply to the broadcast frame to port 48899 with it's IP and SN
It accepts the connection on port 8899 but it does not respond to the data request frame.
So, why? is another contrlCode needed? Should it be a V5 request frame?

I have no TL2 with Embedded Ethernet, so cannot experiment myself...

abc1000, 2017-09-28:
And maybe using a second (local)remote server..?

XtheOne, 2017-09-28:
That is an option, but then you need a listner app.

abc1000, 2017-09-28:
I'v tried nmap (linux based) but nothing special..
I don't know  what else?

XtheOne, 2017-09-28:
I received some documentation. Mainly about how to place the interface inside the inverter.
Is your inverter registered on the omnik portal and can you see production data?
You can try to see the data which is sent by the inverter to the portal by using wireshark.
According to the documentation local logging should work...

abc1000, 2017-09-29:
I see nothing on port 8899, only 80 (my link to webport.)
How do i filter udp, tcp.?
Have now only ipaddress.

Yeah, data logging to omnikportal is allright.

XtheOne, 2017-09-29:
if you run wireshark on windows you only see packets from/to the PC. Bridging with 2 adapters is needed.
Can we continue by email? xtheone (at) huizedegroot (dot) nl

Pietervs2, 2017-09-29:
I had an ethernetmodule installed last week. 
Today I discovered, they've skipped a step after installation.

The ethernetmodule that I have, has 3 lights on it, the one on the left is red, the one on the right is green when a UTP-cable is connected. The one in the middle was off. There's a small button right underneath those three, I pressed that for 3 seconds and the light in the middle came on (also red), after which it started sending data to the Omnikportal again. (In the portal I had to enter the serialnumber of the ethernetcard (1902133xxx), it automatically asked for it after the dataconnection was restored.

But (like others) I'm looking for a working script to upload the data to PVOutput again. Any help with that might me appreciated! :)
(I have attached a file with output I get when I use the command "wget http://192.168.X.XX/js/status.js -q -O - ", hoping that someone can make some sense out of it. I can't find any viable variables in this file :( )
[omnik-output.txt](https://github.com/Woutrrr/Omnik-Data-Logger/files/1345061/omnik-output.txt)

XtheOne, 2017-09-29:
the status.js file is what you also get from the wifi logger. Only you request the js file, not the resulting data from running this js class.
I have no Onmik with a internal ethernet card so cannot test things here...
A capture of data which is sent to the portal can help to see if the frame format is newer than V4 which the WiFi logger uses.

XtheOne, 2017-10-01:
From logging I found that the ethernet logger uses a newer protocol.
Up till now V4 was used and this module uses V5.
I have gotten some documentation but I have no time the comming week to try to implement V5.
As I have no ethernet module testing will be slow as someone else will do the tryouts.

Pietervs2, 2017-10-01:
Hi Michel,

Great you're trying to solve this for us!

If you need any betatesters, don't hesitate to let me know!

Best regards,

Pieter

XtheOne, 2017-10-01:
I have opened an issue under my own repo https://github.com/XtheOne/Inverter-Data-Logger/issues/3
When I get something to test I will let you know.

Pietervs2, 2017-10-12:
> I have opened an issue under my own repo XtheOne/Inverter-Data-Logger#3<https://github.com/XtheOne/Inverter-Data-Logger/issues/3>

Hi Michel,

I see that you have an experimental setup for your script.

I hope I'll be able to test this script this weekend!

Thanks so far!

Best regards,

Pieter

XtheOne, 2017-10-12:
highly experimental as the command frame format is still not known.
If someone can get the firmware out of the GD32F103 MCU with a jtag interface it would greatly help!

Pietervs2, 2017-10-12:
Wil ik wel proberen, maar ik heb geen idee hoe ik dat zou kunnen doen.

Heb je tips hoe ik dat aan kan pakken?

abc1000, 2017-10-12:
I'll think it's a GD32FB07..

abc1000, 2017-10-12:
Do we need a 10p header and a programmer?

XtheOne, 2017-10-12:
It is a GD32F107 MCU and you can read/program it with a ST-link interface.

https://hackaday.com/2013/01/07/reverse-engineering-st-linkv2-firmware/

I have no time this weekend...

XtheOne, 2017-10-12:
Alle tools hier:
http://gd32mcu.21ic.com/en/index

Pietervs2, 2017-10-12:
Ok.

Als ik dit weekend tijd heb ga ik kijken of ik daar iets mee kan :)

Ik ben geen programmeur, dus dit wordt een uitdaging :)

Ik hou je op de hoogte!

M.vr.gr<http://M.vr.gr>.,

Pieter

http://gd32mcu.21ic.com/en/index

XtheOne, 2017-10-13:
De GD32F107RCT6 MCU kun je uitlezen met FlyMcu.exe en een RS232 / 3.3V TTL interface aangesloten op een USART poort. (tenzij het protection bit gezet is...)

mvroosmalen1970, 2018-02-17:
Hi I've been able to capture data send by my converter (lan stick) though unable to decode it.... Does anyone have a clue how to decode?  (same module as reported by :abc1000 commented on 27 Sep 2017)

string captured:
\xa5\\\x00\x10\x02\x00\x14G\xb3eq\x01\x01\x05\xea\x88\x00\x00\x8e\x03\x00\x00\x00\x00\x00\x00\x01\x00\r\x00\x00\x000009600168101010i\x01\xb5\x07\x00\x00\x15\x00\x00\x00\x0e\x00\x00\x00\x00\x00C\t\x00\x00\x00\x00\x87\x13K\x01\x00\x00n\x00\x00\x00X9\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xbe\x0e\x04&\x00\x15\x00\x96\r\xb3\x15

:-)

XtheOne, 2018-05-20:
Hi, I added a server script so that you can let the inverter sent the data.
This should also work with the ethernet module.
See my project.

peterzandbergen, 2019-01-28:
Hi there,

I have been trying to get data from my recently installed inverters. It is an Omnik 3k TL3 with a serial number of 640xxxx. I can send the request command v4 but do not get a response. I assume that the protocol has changed to a new version. Can you help me in finding info on the new protocol, or how I can experiment and find what command I need to send.

Thanks

XtheOne, 2019-01-28:
I am trying to get my script to work with the newer V5 protocol.
I have some info but little help from Omnik /iGEN tech.

peterzandbergen, 2019-01-28:
So you say I need to use the V5 protocol for this wifi module. 

Does it help if I give you the firmware version? Or any other info I can help you with?

mivale, 2019-05-09:
For anyone looking for a simple datalogger -> pvoutput script i've written a simple bash script which works with an external Omnik Wifi Kit S-WE02. Pieced it together from all the info found here and elsewhere. Just giving back :-)

#!/bin/bah
#
today=`date '+%Y%m%d'`
minute=`date '+%H:%M'`

url="http://192.168.1.17/js/status.js"

# get most recent webdata from Hosola / Omnik inverter
content=$(curl --connect-timeout 30 -s $url | tr ';' '\n' | grep -e "^myDeviceArray\[0\]" | sed -e 's/"//g' | sed 's/myDeviceArray\[0\] = //')

if [[ -z "$content" ]]; then
    exit 1
fi

# get current power value, put all available values in array
set -- "$content"
IFS=","; declare -a Array=($*)

# NLDNXXXXXXXXXXXX,NL1-V1.0-XXXX-4,V2.0-XXXX,omnikXXXXtl ,X000,1070,790,160839,,1,

# please change with correct sid and key from pvoutput.org account
auth="sid=<SID>&key=<KEY_HEX>"

# post power value to pvoutput site
postdatastring="${auth}&v2=${Array[6]}&t=$minute&d=$today"

echo -n "$today $minute " >> ~/omnik/output.log
curl -s "http://pvoutput.org/service/r2/addstatus.jsp?$postdatastring" >> ~/omnik/output.log
echo "" >> ~/omnik/output.log

##################

Running it on a Raspberry Pi

mainframecn, 2019-09-16:
Thanks mivale, i used your information to create a sensor in HA with PV output.