Woutrrr/Omnik-Data-Logger #28: Omnik no data/ several errors

pgas37, 2017-12-13:
I followed the instructions to set up de Omnik, but no data and errors. Omnik Vrtual and sensors are visible.

Livestat.py debug gives a error:
>>> %Debug LiveStats.py
Traceback (most recent call last):
  File "/home/pi/Omnik-Data-Logger/LiveStats.py", line 7, in <module>
    import OmnikExport
  File "/home/pi/Omnik-Data-Logger/OmnikExport.py", line 11, in <module>
    import ConfigParser
ImportError: No module named 'ConfigParser
========================================
InverterMsg.py debug gives an error:
Python 3.4.2 (/usr/bin/python3)
>>> %Run InverterMsg.py
  File "/home/pi/Omnik-Data-Logger/InverterMsg.py", line 73
    print self.__get_short(59)
             ^
SyntaxError: invalid syntax
===========================================
config looks like this:
general]
# General:enabled_plugins
# Choose which outputs to use
# Possible options: MysqlOutput,PVoutputOutput,ConsoleOutput,CSVOutput
enabled_plugins = type:DomoticzOutput

[inverter]
# IP address of your Omnik inverter
ip = 10.10.xxx.xxx.    # is ipades visible in Omnik GUI. # 192.168.xxx.xxx = normal ipadres DCHP router   ##  ??????
# Default for a Omnik with Wifi module
port = 8899
# S/N of the wifi kit
wifi_sn = 603080xxx   # fill out s/n
#use temperature of inverter for pvoutput
use_temperature = true

[mysql]
# Host where the mysql server is active
host = 127.0.0.1
user =
pass =
database =

[pvout]
# These two can be found at http://pvoutput.org/account.jsp
#apikey = NOTAREALAPIKEY86e2258d4e29169fb79cf18b00
#sysid  = 12345

[csv]
disable_header = false

[log]
# Log:Output
# Possible options: none,console,file (combinations are possible)
# Use none to disable logging
type = console

# Log:level
# Possible options: critical, error, warning, info, debug
level = debug

# Log:filename
# Output file for file logger
filename = omnik-export.log

[domoticz]
domoticz_host      = 192.168.xxx.xxx #<IP OF DOMOTICZ MACHINE>
domoticz_port      = 8080 #<Port of DOMOTICZ MACHINE>
domoticz_url       = json.htm

# Provide IDX here of the specific devices
idx_Temp      = 204
idx_PV1_U     = 205
idx_PV2_U     = 206
idx_PV1_A     = 207
idx_PV2_A     = 208
idx_AC_Output = 209
idx_E_Total   = 210
idx_E_Current = 211
==================================

I use a Omnik wifi module starting s/n 603xx
Domoticz version 3.8153
Rapsberrypi. V3

Please help?

thanks
Paul

XtheOne, 2017-12-13:
Looks like you use python 3.x
This script was written under 2.7
It looks it needs updating to run under 3.x

pgas37, 2017-12-14:
Michel,

Thanks for your info.
I think i have to wait for an update...... from the author?

Regards
Paul

==>iPadp

XtheOne, 2017-12-14:
I had a quick look, it is some work to make the scripts Python 2 and 3 compatible.
Will schedule this.

pgas37, 2017-12-14:
I use a raspberrypi with raspbian-jesse. I do not know how to downgrade python and what consequences a downgrade has on the system and the use of domoticz and/or other software.
You wrote that you will schedule this in your own project “inverterExport” . I am new here so i do not know where i can find your “ inverterproject” . Is this a part of the https://github.com/Woutrrr/xxx or is this something different?
Please put me on the right track...
Thanks

XtheOne, 2017-12-14:
https://github.com/XtheOne/Inverter-Data-Logger

pgas37, 2018-02-18:
Michel,
Can you tell me if the scripts are already python 3 compatibel?
thanks
paul

XtheOne, 2018-03-17:
I have little time, but worked on them and livestats works now.

pgas37, 2018-03-18:
thanks for your time

I reinstall following the instructions. No data.........
In the meantime i had to replace  miy wifi module s/n start with a 1  : 1604573xxx
I do not know wether the leading 1 is the wrongdoer or something else.
too much dependencies which cann't be checked to get this working

regards
Paul

JPBTEW, 2018-08-03:
I'm also having problems with this script... can't figure it out....
Livestats works well and omnikexport is giving the following error...

pi@raspberrypitest:~/Omnik-Data-Logger $ python OmnikExport.py
2018-08-03 20:36:03,464 DEBUG Importing output plugin ConsoleOutput
2018-08-03 20:36:03,466 DEBUG Importing output plugin DomoticzOutput
2018-08-03 20:36:03,524 INFO connecting to 192.168.1.117 port 8899
2018-08-03 20:36:04,163 INFO ID: NLDN402014481122
2018-08-03 20:36:04,164 DEBUG Run pluginConsoleOutput
ID: NLDN402014481122
E Today: 21.24   Total: 13364.4
H Total: 14652   Temp:  41.5 
PV1   V: 214.9   I:  0.4
PV2   V: 227.9   I:  0.6
PV3   V:    -1   I:   -1
L1    P:   203   V: 235.2   I:  0.8   F: 50.04
L2    P:    -1   V:    -1   I:   -1   F:    -1
L3    P:    -1   V:    -1   I:   -1   F:    -1
2018-08-03 20:36:04,166 DEBUG Run pluginDomoticzOutput
Traceback (most recent call last):
  File "OmnikExport.py", line 170, in <module>
    omnik_exporter.run()
  File "OmnikExport.py", line 82, in run
    plugin.process_message(msg)
  File "/home/pi/Omnik-Data-Logger/outputs/DomoticzOutput.py", line 46, in process_message
    urllib.urlopen(full_url)
  File "/usr/lib/python2.7/urllib.py", line 87, in urlopen
    return opener.open(url)
  File "/usr/lib/python2.7/urllib.py", line 213, in open
    return getattr(self, name)(url)
  File "/usr/lib/python2.7/urllib.py", line 364, in open_http
    return self.http_error(url, fp, errcode, errmsg, headers)
  File "/usr/lib/python2.7/urllib.py", line 377, in http_error
    result = method(url, fp, errcode, errmsg, headers)
  File "/usr/lib/python2.7/urllib.py", line 689, in http_error_401
    errcode, errmsg, headers)
  File "/usr/lib/python2.7/urllib.py", line 386, in http_error_default
    raise IOError, ('http error', errcode, errmsg, headers)
IOError: ('http error', 401, 'Unauthorized', <httplib.HTTPMessage instance at 0x767bbb48>)

I have no idea what i am doing wrong.....

can someone help me????