Woutrrr/Omnik-Data-Logger #30: Power value L3

Schampbakken, 2018-02-18:
Hello,

i'm missing the power value of L3. L1 and 2 work perfect but L3 is missing.
When I look in InverterMsg.py I see the next code.
I added a print for debug

``
    def p_ac(self, i=1):
        """Power output of the output channel

        Available channels are 1, 2 or 3; if no tin this range the function will
        default to channel 1.

        Args:
            i (int): output channel (valid values: 1, 2, 3)

        Returns:
            float: Power output of channel i
        """
        if i not in range(1, 4):
            i = 1
        num = 59 + (i - 1) * 4
        print num, ": L", i, ": ", int(self.__get_short(num, 1))
        return int(self.__get_short(num, 1))  # Don't divide
``
In the console it looks like:
59 : L 1 :  394
63 : L 2 :  380
67 : L 3 :  0

What can this be? Or is this not the power for each phase?

Schampbakken, 2018-02-18:
Think I found it!
I think the multilpier should be 2 instead of 4.
Think the value is in 2 bytes instead of 4.

`num = 59 + (i - 1) * 2`

XtheOne, 2018-03-19:
Nope... From the data frame layout table:
f_ac1           57       2     100      Hz     Frequency of the AC output channel 1.
p_ac1           59       2       1      W      Power output of the AC output channel 1.
f_ac2           62       2     100      Hz     Frequency of the AC output channel 2.
p_ac2           63       2       1      W      Power output of the AC output channel 2.
f_ac3           65       2     100      Hz     Frequency of the AC output channel 3.
p_ac3           67       2       1      W      Power output of the AC output channel 3.
Maybe you have an old logger which outputs an old frame type.
A dump is needed then to look.