Woutrrr/Omnik-Data-Logger #35: Send attributes to MQTT [ Integration HomeAssistant ]

RazorCopter, 2018-09-11:
# NOTE
# Sending attributes with mqtt paho client for integration with HomeAssistant
# Need Paho Client - > pip install paho-mqtt
# Create new file MQTTOutput.py to /user/Omnik-Data-Logger/outputs
# Edit config.cfg -> enabled_plugins = MQTTOutput
######################
# Create new sensor:
######## sensor.yaml ########
# - platform: mqtt
#   name: "Energia Solare Oggi"
#  state_topic: "tele/inverter/SENSOR"
#  value_template: "{{value_json['ENERGY'].E_Today }}"
#  unit_of_measurement : "kWh"
#  expire_after: 120
#  .......
#################################################
############ MQTTOutput.py ######### ##############
import os
import PluginLoader
import paho.mqtt.client as mqtt
import json
import datetime

# Define Variables
MQTT_HOST = "192.168.3.10"
MQTT_PORT = 1883
MQTT_USER = "yourUser"
MQTT_PASS = "yourPassword"
MQTT_KEEPALIVE_INTERVAL = 45
MQTT_TOPIC = "tele/inverter/SENSOR"

#######################################################################
class ConsoleOutput(PluginLoader.Plugin):
    """Outputs the data from the Omnik inverter to stdout"""
    def process_message(self, msg):
        """Output the information from the inverter to stdout.
        Args:
            msg (InverterMsg.InverterMsg): Message to process
        """
        actual_time = datetime.datetime.now().replace(microsecond=0).isoformat()

        json_body={"Time":actual_time,"ENERGY":{"Temp":(msg.temperature),"H_Total":(msg.h_total),"E_Today":(msg.e_today),"E_Total":(msg.e_total),"VPV1":(msg.v_pv(1)),"$
        mqttc = mqtt.Client("inverter")
        mqttc.username_pw_set(username=MQTT_USER, password=MQTT_PASS)
        mqttc.connect(MQTT_HOST, MQTT_PORT, MQTT_KEEPALIVE_INTERVAL)
        mqtt_msg=json.dumps(json_body)
        mqttc.publish(MQTT_TOPIC, mqtt_msg)
        mqttc.disconnect()

eterpstra, 2019-03-22:
Hi, RazorCopter 

I think this line is not correct:

        `json_body={"Time":actual_time,"ENERGY":{"Temp":(msg.temperature),"H_Total":(msg.h_total),"E_Today":(msg.e_today),"E_Total":(msg.e_total),"VPV1":(msg.v_pv(1)),"$`
it ends with a $ but i guess it needs to be:

`json_body={"Time":actual_time,"ENERGY":{"Temp":(msg.temperature),"H_Total":(msg.h_total),"E_Today":(msg.e_today),"E_Total":(msg.e_total),"VPV1":(msg.v_pv(1))}}`

THis worked for me. But is it also possible to have E-Instant / Real-time information?

RazorCopter, 2019-03-22:
sorry, copy / paste error!  :)
https://pastebin.com/qD51X299
It is not possible to have data in real time because it is not possible to interrogate the inverter but only to listen.