Woutrrr/Omnik-Data-Logger #36: Omnik S/N starting with 641 and no data from inverter

JRie3233, 2019-03-02:
This week received a new inverter with a serial number started with 641.

After update the config file with IP and serial and the output the following error 
pi@raspberrypi3:~/inverter $ python LiveStats.py
2019-03-02 11:12:01,513 DEBUG Importing output plugin ConsoleOutput
2019-03-02 11:12:01,516 INFO connecting to 192.168.1.145 port 8899
Traceback (most recent call last):
  File "LiveStats.py", line 17, in <module>
    omnik_exporter.run()
  File "/home/pi/inverter/OmnikExport.py", line 88, in run
    data = inverter_socket.recv(1024)
socket.timeout: timed out

No idea what goes wrong ?

eterpstra, 2019-03-21:
I have a simalar issue with my Omrik inverter.

019-03-21 09:51:25,353 DEBUG Importing output plugin
Traceback (most recent call last):
  File "OmnikExport.py", line 170, in <module>
    omnik_exporter.run()
  File "OmnikExport.py", line 52, in run
    __import__(plugin_name)
ValueError: Empty module name
user@user-MacBookPro:~/Omnik-Data-Logger$ crontab -e
crontab: installing new crontab
user@user-MacBookPro:~/Omnik-Data-Logger$ python OmnikExport.py
2019-03-21 09:53:38,402 DEBUG Importing output plugin
Traceback (most recent call last):
  File "OmnikExport.py", line 170, in <module>
    omnik_exporter.run()
  File "OmnikExport.py", line 52, in run
    __import__(plugin_name)
ValueError: Empty module name

Livestats.py is working fine...

gniersma, 2019-05-27:
Same problem here, used you script for a while, had to replace my inverter no with ser no: 640058498. 
Both LiveStats.py and OmnikExport.py doesn't work with this new wifi module. Suggestions are welcome

pi@raspberrypi:~/Omnik-Data-Logger $ python LiveStats.py
2019-05-27 09:21:02,816 DEBUG Importing output plugin ConsoleOutput
2019-05-27 09:21:03,069 INFO connecting to 192.168.xx.59 port 8899
Traceback (most recent call last):
  File "LiveStats.py", line 17, in <module>
    omnik_exporter.run()
  File "/home/pi/Omnik-Data-Logger/OmnikExport.py", line 72, in run
    inverter_socket.sendall(OmnikExport.generate_string(wifi_serial))
  File "/home/pi/Omnik-Data-Logger/OmnikExport.py", line 160, in generate_string
    reversed(range(0, len(double_hex), 2))]
  File "/usr/lib/python2.7/encodings/hex_codec.py", line 42, in hex_decode
    output = binascii.a2b_hex(input)
TypeError: Non-hexadecimal digit found

pi@raspberrypi:~/Omnik-Data-Logger $ python OmnikExport.py
2019-05-27 09:23:01,160 DEBUG Importing output plugin DomoticzOutput
2019-05-27 09:23:01,226 INFO connecting to 192.168.xx.59 port 8899
Traceback (most recent call last):
  File "OmnikExport.py", line 170, in <module>
    omnik_exporter.run()
  File "OmnikExport.py", line 72, in run
    inverter_socket.sendall(OmnikExport.generate_string(wifi_serial))
  File "OmnikExport.py", line 160, in generate_string
    reversed(range(0, len(double_hex), 2))]
  File "/usr/lib/python2.7/encodings/hex_codec.py", line 42, in hex_decode
    output = binascii.a2b_hex(input)
TypeError: Non-hexadecimal digit found

stingone, 2019-06-18:
Same here with S/N starting with 6372 any idea to solve 

File "OmnikExport.py", line 170, in <module>
    omnik_exporter.run()
  File "OmnikExport.py", line 73, in run
    data = inverter_socket.recv(1024)
socket.timeout: timed out

stingone, 2019-06-18:
> This week received a new inverter with a serial number started with 641.
> 
> After update the config file with IP and serial and the output the following error
> pi@raspberrypi3:~/inverter $ python LiveStats.py
> 2019-03-02 11:12:01,513 DEBUG Importing output plugin ConsoleOutput
> 2019-03-02 11:12:01,516 INFO connecting to 192.168.1.145 port 8899
> Traceback (most recent call last):
> File "LiveStats.py", line 17, in 
> omnik_exporter.run()
> File "/home/pi/inverter/OmnikExport.py", line 88, in run
> data = inverter_socket.recv(1024)
> socket.timeout: timed out
> 
> No idea what goes wrong ?

Did you found a solution?

Jobvanr, 2019-07-07:
Same issue for me.. 

python LiveStats.py
2019-07-07 19:15:40,503 DEBUG Importing output plugin ConsoleOutput
2019-07-07 19:15:40,504 INFO Connecting to logger with IP: 192.168.2.7 and SN 641060337
2019-07-07 19:15:40,505 INFO connecting to 192.168.2.7 port 8899
2019-07-07 19:15:40,511 DEBUG RAW sent Packet (len=16): 68:02:41:b1:f1:cd:35:26:f1:cd:35:26:01:00:27:16  hA5&5&'
2019-07-07 19:15:43,514 ERROR Timeout connecting to logger with IP: 192.168.2.7 and SN 641060337, trying next logger.`

sincze, 2020-04-11:
Maybe this can help?
https://github.com/Woutrrr/Omnik-Data-Logger/issues/42#issuecomment-603436447

elacour, 2020-05-09:
I face the same problem, does anyone found the tcp request for SN 643xxxxxx ? Status page lacks many usefull data.

sincze, 2020-05-10:
Maybe a second listener on your network might help to obtain idd info like V / A e.g.? [link](http://www.mb200d.nl/wordpress/2015/11/omniksol-4k-tl-wifi-kit/)

elacour, 2020-05-18:
I was able to change the IP address of omni server a and use mine, using https://github.com/XtheOne/Inverter-Data-Logger.git in server listen mode. Thought data received are pretty wrong  :(

ojeysky, 2021-01-27:
I am having exactly this same issue anyone got through this.
`/Inverter-Data-Logger $ sudo python LiveStats.py 
2021-01-27 09:25:11,080 DEBUG Importing output plugin ConsoleOutput
2021-01-27 09:25:14,190 INFO Loggers found  on the network: ['192.168.0.180', '17594xxxx'].
2021-01-27 09:25:14,191 INFO Connecting to logger with IP: 192.168.0.180 and SN 17594xxxx
2021-01-27 09:25:14,191 INFO connecting to 192.168.0.180 port 8899
2021-01-27 09:25:14,267 DEBUG RAW sent Packet (len=16): 68:02:41:b1:39:69:df:68:39:69:df:68:01:00:c7:16  hA9ih9ih
2021-01-27 09:25:17,272 ERROR Timeout connecting to logger with IP: 192.168.0.180 and SN 1759471929, trying next logger.
`