Woutrrr/Omnik-Data-Logger #37: Omnik Datalogger S/N starting with 626 and not able to connect

lepirlouit, 2019-03-26:
Hi,
I just installed solar panels with an omnik inverter.
The datalogger is starting with 626
open ports are 80 and 8899
I'm not able to connect on port 8899 with your script
I can setup a bridge to be able to capture traffic.
Which software do I need to generate data to be captured ?
Currently I only see packets going to server 47.88.8.200 (data1.solarmanpv.com) to port 10000
Can we get data out of this to be able to connect to port 8899 on the inverter ?

dondipietro, 2020-04-11:
Hello,
Just to tell you I have exactly the same problem :
My Wifi is igen and serial is 626xxxxxx

2020-04-11 10:06:22,509 DEBUG Importing output plugin DomoticzOutput
2020-04-11 10:06:22,563 INFO connecting to 192.168.1.14 port 8899
Traceback (most recent call last):
  File "OmnikExport.py", line 171, in <module>
    omnik_exporter.run()
  File "OmnikExport.py", line 74, in run
    data = inverter_socket.recv(1024)
socket.timeout: timed out

Do you find any solution ?

sincze, 2020-04-11:
Maybe this can help?
https://github.com/Woutrrr/Omnik-Data-Logger/issues/42#issuecomment-603436447