Woutrrr/Omnik-Data-Logger #40: Python errors to Domoticz

keitje, 2019-05-19:
Hello, I tried to follow the wiki from Domoticz (https://www.domoticz.com/wiki/Omnik_Solar_Inverter). I get data out of my Omnik, but I'm struggling with the DomoticzOutput.py which was created. Everything is setup as mentioned in the wiki but I get Python errors when running OmnikExport.py.

Errors are:
2019-05-19 11:30:40,837 DEBUG Importing output plugin DomoticzOutput
Traceback (most recent call last):
  File "OmnikExport.py", line 170, in <module>
    omnik_exporter.run()
  File "OmnikExport.py", line 52, in run
    __import__(plugin_name)
  File "/home/pi/Omnik-Data-Logger/outputs/DomoticzOutput.py", line 30
    def process_message(self, msg):
    ^
IndentationError: unexpected indent

I have to mention, I cannot read a single line of Python... (sorry), the error mentions Unexpected indent, but I dont know where of what to look for. I'm running Python 2.7.
Can someone help me to the right direction? Also, do I need to run Python 3 or is 2.7 still needed?

basssment, 2020-06-29:
Same problem here, followed the wiki.
I do get the output correct when i run LiveStats.py but when i run OmnikExport.py i get:
2020-06-28 20:22:17,341 DEBUG Importing output plugin
Traceback (most recent call last):
  File "OmnikExport.py", line 170, in <module>
    omnik_exporter.run()
  File "OmnikExport.py", line 52, in run
    __import__(plugin_name)
ValueError: Empty module name

clmcavaney, 2020-07-27:
Do you have any plugins enabled in your config.cfg file?

For example:
    [general]
    # General:enabled_plugins
    # Choose which outputs to use
    # Possible options: MysqlOutput,PVoutputOutput,ConsoleOutput,CSVOutput
    enabled_plugins = ConsoleOutput