Woutrrr/Omnik-Data-Logger #42: Issues Omnik inverter with SN: 80911****

FlorisHigh3IT, 2020-03-23:
I used to read the data with this script for showing in my Node Red Dashboard. Worked perfect!

After my Omnik inverter died, this week I got a new Omnik2500tl. Unfortunately now it does not work anymore. Any idea how to solve this?

This is the result on LiveStats.py:

python LiveStats.py
2020-03-15 13:15:38,539 DEBUG Importing output plugin ConsoleOutput
2020-03-15 13:15:41,693 INFO Loggers found on the network: ['192.168.0.76', '80 9110293'].
2020-03-15 13:15:41,693 INFO Connecting to logger with IP: 192.168.0.76 and SN 8 0911****
2020-03-15 13:15:41,694 INFO connecting to 192.168.0.76 port 8899
2020-03-15 13:15:41,696 DEBUG RAW sent Packet (len=16): 68:02:41:b1:15:0b:3a:30: 15:0b:3a:30:01:00:09:16 hA:0:0
2020-03-15 13:15:44,699 ERROR Timeout connecting to logger with IP: 192.168.0.76 and SN 80911****, trying next logger.

sincze, 2020-03-23:
I think this has something to do with the serial number. Do you still have the old data logger SN *6**** ? In my memory the script only works with those. Else you could scrape the data from the inverter itself. 
`var myDeviceArray=new Array(); myDeviceArray[0]="AANN3020,V5.04Build230,V4.13Build253,Omnik3000tl,3000,1313,685,9429,,1,";`

If you can open the http://inverter-ip/js/status.js

FlorisHigh3IT, 2020-03-24:
No, they changed the whole inverter including the datalogger. Unfortunately the status.js returns a 404 on the new datalogger :(

sincze, 2020-03-24:
Can you open the inverter page.
F12 in Chrome on (developer mode) to see what the page does to retrieve the
data?

FlorisHigh3IT, 2020-03-24:
My bad, I tried the http://inverter-ip/js/status.js and it does return information!

It starts with
```
var myDeviceArray=new Array();
myDeviceArray[0]="AARN252019814151,V6.00Build260,V4.13Build262,Omnik2500tl ,2500,47,1251,1084,,1,";
```

Any example script to start from?

FlorisHigh3IT, 2020-03-24:
Wait, maybe I should try this plugin: https://github.com/sincze/Domoticz-Omnik-Local-Web-Plugin/tree/master

sincze, 2020-03-24:
Give it a go.
I build it.

FlorisHigh3IT, 2020-03-25:
Works like a charm! Some small questions but I will post these on the Local Web Plugin Git.

Thanks for your help and effort!

closed 2020-03-25 by FlorisHigh3IT